XCSoar/XCSoar #994: Android CI build pipeline is broken

hjr, 2022-08-18:
Deploy to staging server: fails with tha following last log lines:
Run easingthemes/ssh-deploy@main
[general] GITHUB_WORKSPACE:  /home/runner/work/XCSoar/XCSoar
⚠️ [INPUTS] SSH_PRIVATE_KEY is mandatory
⚠️ [INPUTS] REMOTE_HOST is mandatory
⚠️ [INPUTS] REMOTE_USER is mandatory
⚠️ [INPUTS] Inputs not valid, aborting ...
 1: 0xa1a640 node::Abort() [/home/runner/runners/2.295.0/externals/node12/bin/node]
 2: 0xa90649  [/home/runner/runners/2.295.0/externals/node12/bin/node]
 3: 0xc06599  [/home/runner/runners/2.295.0/externals/node12/bin/node]
 4: 0xc08387 v8::internal::Builtin_HandleApiCall(int, unsigned long*, v8::internal::Isolate*) [/home/runner/runners/2.295.0/externals/node12/bin/node]
 5: 0x140dd[19](https://github.com/XCSoar/XCSoar/runs/7904175280?check_suite_focus=true#step:10:20)  [/home/runner/runners/2.295.0/externals/node12/bin/node]

lordfolken, 2022-08-18:
https://github.com/XCSoar/XCSoar/commit/feea2fc72ed1d44f5e34ac083db120c8c12c6dae fixes this. So PR shouldn't try to upload any more.

lordfolken, 2022-08-18:
I'm removing the breaking tag, as this is meant for the application itself.

hjr, 2022-08-18:
Saw your change, but did not see it fixing the issue. The PR to https://github.com/XCSoar/XCSoar/commit/feea2fc72ed1d44f5e34ac083db120c8c12c6dae still showed the same issue?!

hjr, 2022-08-18:
All PR's still show the same failure on Android. The CI pipeline is not yet fixed.

closed 2022-08-23 by MaxKellermann