XCSoar/XCSoar #998: build target KOBO failing, thirdparty.mk:41

ubx, 2022-08-20:
Reproducer:

1. git clone --recurse-submodules https://github.com/XCSoar/XCSoar
2. cd XCSoar
3. docker build --no-cache --file ide/docker/Dockerfile -t xcsoar/xcsoar-build:latest ./ide/
4. docker run --mount type=bind,source="$(pwd)",target=/opt/xcsoar -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile KOBO

```
make[1]: *** [Makefile:1004: all] Error 2
Traceback (most recent call last):
  File "/opt/xcsoar/./build/thirdparty.py", line 210, in <module>
    x.build(toolchain)
  File "/opt/xcsoar/build/python/build/project.py", line 80, in build
    self._build(toolchain)
  File "/opt/xcsoar/build/python/build/gcc.py", line 8, in _build
    AutotoolsProject._build(self, toolchain.native, target_toolchain=toolchain)
  File "/opt/xcsoar/build/python/build/autotools.py", line 114, in _build
    self.build_make(toolchain, build)
  File "/opt/xcsoar/build/python/build/makeproject.py", line 31, in build_make
    self.make(toolchain, wd, self.get_make_args(toolchain))
  File "/opt/xcsoar/build/python/build/makeproject.py", line 27, in make
    subprocess.check_call(['make'] + args,
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '--quiet', '-j32']' returned non-zero exit status 2.
make: *** [build/thirdparty.mk:41: output/KOBO/lib/stamp] Error 1
rm output/KOBO/dbg/src/Device/Driver/../dirstamp
```

MaxKellermann, 2022-08-22:
You omitted the important part of the error message.

ubx, 2022-08-23:
The error message:
```
checking for string.h... /opt/xcsoar/output/src/binutils-2.39/missing: 81: makeinfo: not found
WARNING: 'makeinfo' is missing on your system.
         You should only need it if you modified a '.texi' file, or
         any other file indirectly affecting the aspect of the manual.
         You might want to install the Texinfo package:
         <http://www.gnu.org/software/texinfo/>
         The spurious makeinfo call might also be the consequence of
         using a buggy 'make' (AIX, DU, IRIX), in which case you might
         want to install GNU make:
         <http://www.gnu.org/software/make/>
yes
make[4]: *** [Makefile:1772: doc/bfd.info] Error 127
make[3]: *** [Makefile:1932: info-recursive] Error 1
make[2]: *** [Makefile:3056: all-bfd] Error 2
```

lordfolken, 2022-08-23:
That's  only a warning, so it shouldn't abort compilation. I'll try tonight.

MaxKellermann, 2022-08-23:
There's still the important part of the error message missing.

ubx, 2022-08-23:
I'm confused. Maybe this is the reason:
```
checking for ANSI C header files... /opt/xcsoar/output/src/binutils-2.39/missing: 81: makeinfo: not found
WARNING: 'makeinfo' is missing on your system.
         You should only need it if you modified a '.texi' file, or
         any other file indirectly affecting the aspect of the manual.
         You might want to install the Texinfo package:
         <http://www.gnu.org/software/texinfo/>
         The spurious makeinfo call might also be the consequence of
         using a buggy 'make' (AIX, DU, IRIX), in which case you might
         want to install GNU make:
         <http://www.gnu.org/software/make/>
make[4]: *** [Makefile:1772: doc/bfd.info] Error 127
make[3]: *** [Makefile:1932: info-recursive] Error 1
make[2]: *** [Makefile:3056: all-bfd] Error 2
```

You can see the whole log here [console.log](https://github.com/XCSoar/XCSoar/files/9404180/console.log)

lordfolken, 2022-08-23:
adding makeinfo to the (texinfo package) to the container actually fixes the compile process...

PR #1007 (2022-08-23): install-debian-packages: add missing texinfo package
fixes #998
closed 2022-08-23 by MaxKellermann