XCSoar/XCSoar #999: build target ANDROID failing: error: use of undeclared identifier 'default_events'

iltis42, 2022-08-20:
How to reproduce: Freshly pull XCSoar, optionally build docker image (failure happens also without building it), and trigger the ANDROID build command. It builds a while hence runs into a compiler error at file InputDefaults.cpp that cannot resolve identifier "default_events" (1).  
What maybe causing the problem IMHO: The default_events are given as function arguments, hence in ARRAY_SIZE those should be globally defined on some way, maybe this was intended by #include ImputEvent_default.cpp in there, hence this file is obviously missing in repo (3).

1)
git clone --recursive https://github.com/XCSoar/XCSoar
cd XCSoar
[ sudo docker build   --file ide/docker/Dockerfile     -t xcsoar/xcsoar-build:latest ./ide/ ]
sudo docker run     --mount type=bind,source="$(pwd)",target=/opt/xcsoar     -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile ANDROID
make[1]: Entering directory '/opt/xcsoar'
make[1]: Entering directory '/opt/xcsoar'
make[1]: Entering directory '/opt/xcsoar'
make[1]: Entering directory '/opt/xcsoar'
  CXX     output/ANDROID/x86_64/dbg/src/Device/Driver.o
  CXX     output/ANDROID/arm64-v8a/dbg/src/Device/Driver.o
  CXX     output/ANDROID/armeabi-v7a/dbg/src/Device/Driver.o
  CXX     output/ANDROID/x86/dbg/src/Device/Driver.o
  CXX     output/ANDROID/arm64-v8a/dbg/src/Input/InputDefaults.o
  CXX     output/ANDROID/x86_64/dbg/src/Input/InputDefaults.o
  CXX     output/ANDROID/armeabi-v7a/dbg/src/Input/InputDefaults.o
  CXX     output/ANDROID/x86/dbg/src/Input/InputDefaults.o
src/Input/InputDefaults.cpp:152:29: error: use of undeclared identifier 'default_events'
                 ARRAY_SIZE(default_events),
                            ^
src/Input/InputDefaults.cpp:152:29: error: use of undeclared identifier 'default_events'
                 ARRAY_SIZE(default_events),
                            ^
src/Input/InputDefaults.cpp:152:29: error: use of undeclared identifier 'default_events'
src/Input/InputDefaults.cpp:152:29: error: use of undeclared identifier 'default_events'
src/Input/InputDefaults.cpp:152:29: error: use of undeclared identifier 'default_events'
                 ARRAY_SIZE(default_events),
                            ^
src/Input/InputDefaults.cpp:152:29: error: use of undeclared identifier 'default_events'
src/Input/InputDefaults.cpp:150:18: error: use of undeclared identifier 'default_modes'
                 default_modes,
                 ^
src/Input/InputDefaults.cpp:151:18: error: use of undeclared identifier 'default_events'
                 default_events,
                 ^
src/Input/InputDefaults.cpp:153:18: error: use of undeclared identifier 'default_gesture2event'
                 default_gesture2event,
                 ^
src/Input/InputDefaults.cpp:154:18: error: use of undeclared identifier 'default_key2event'
                 default_key2event, default_gc2event, default_n2event,
                 ^
src/Input/InputDefaults.cpp:154:37: error: use of undeclared identifier 'default_gc2event'
                 default_key2event, default_gc2event, default_n2event,
                                    ^
src/Input/InputDefaults.cpp:154:55: error: use of undeclared identifier 'default_n2event'
                 default_key2event, default_gc2event, default_n2event,
                                                      ^
src/Input/InputDefaults.cpp:155:18:src/Input/InputDefaults.cpp:150:18: error:  use of undeclared identifier 'default_modes'
                 default_modes,
                 ^
src/Input/InputDefaults.cpperror: :151:18: error: use of undeclared identifier 'default_events'
                 default_events,
                 ^
src/Input/InputDefaults.cpp:153:18: error: use of undeclared identifier 'default_gesture2event'
                 default_gesture2event,
                 ^
src/Input/InputDefaults.cpp:154:18: error: use of undeclared identifier 'default_key2event'
                 default_key2event, default_gc2event, default_n2event,
                 ^
src/Input/InputDefaults.cpp:154:37: error: use of undeclared identifier 'default_gc2event'
                 default_key2event, default_gc2event, default_n2event,
                                    ^
src/Input/InputDefaults.cpp:154use of undeclared identifier 'default_labels'
:55:                  default_labels);
                 ^

2)
root@2340d16acd11:/opt/xcsoar/src# find . -name "*" -type f |xargs grep default_events
./Input/InputDefaults.cpp:               const InputConfig::Event *default_events,
./Input/InputDefaults.cpp:               unsigned num_default_events,
./Input/InputDefaults.cpp:  assert(num_default_events <= InputConfig::MAX_EVENTS);
./Input/InputDefaults.cpp:  input_config.events.resize(num_default_events + 1);
./Input/InputDefaults.cpp:  std::copy_n(default_events, num_default_events,
./Input/InputDefaults.cpp:                 default_events,
./Input/InputDefaults.cpp:                 ARRAY_SIZE(default_events),

3)
root@2340d16acd11:/opt/xcsoar/src# find . -name   InputEvent_default.cpp
root@2340d16acd11:/opt/xcsoar/src#

ubx, 2022-08-20:
I'm able to build target ANDROID with:
```
git clone --recurse-submodules https://github.com/XCSoar/XCSoar
cd XCSoar
docker build --no-cache --file ide/docker/Dockerfile -t xcsoar/xcsoar-build:latest ./ide/
docker run --mount type=bind,source="$(pwd)",target=/opt/xcsoar -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile ANDROID
```

iltis42, 2022-08-20:
Ok, many thanks for the above command, i have just retriggered with slightly modified clone command and omitting the cache.

iltis42, 2022-08-20:
Okay, .apk is there, thanks for the command, guess this is worth to do some adjust in manual ?
 
AR      output/ANDROID/arm64-v8a/dbg/math.a
  LINK    output/ANDROID/arm64-v8a/dbg/bin/libxcsoar-ns.so
  STRIP   output/ANDROID/arm64-v8a/dbg/bin/libxcsoar.so
rm output/ANDROID/arm64-v8a/dbg/src/Device/Driver/../dirstamp
make[1]: Leaving directory '/opt/xcsoar'
  APK     output/ANDROID/dbg/build/unsigned.apk
  ALIGN   output/ANDROID/dbg/build/aligned.apk
  SIGN    output/ANDROID/bin/XCSoar-debug.apk
rm output/host/tools/GenerateSineTables/../dirstampmake: unlink: output/host/tools/GenerateSineTables/../dirstamp: Not a directory
 output/host/tools/GenerateSineTables.o

closed 2022-08-20 by iltis42