XCSoar/XCSoar #1002: Support for native serial ports on Android

iltis42, 2022-08-21:
Recent build of XCSoar version 7.24 with latest changeset.
--------------
What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
As also discussed in the thread below (at wrong construction site under a topic concerning UsbSerial), currently i am missing any support for native serial ports on Android:
https://github.com/XCSoar/XCSoar/issues/968?fbclid=IwAR0Cyd3nC0qLTNc7GRGn70yROM2i6A9uNzyUhf0RXTkRZJHk5NnoqEIh9pU#issuecomment-1221535391

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
I want to make use native serial ports that are not connected to an USB port on an Android board that provides 3 of them. 
A demo program with this library can show those ports well on that board.
My current idea is to to add support for native serial ports on Android by integrating "Android-Serialport", e.g. from here
https://github.com/xmaihh/Android-Serialport
what comes with a description and integrate this as third party lib into XCSoar like for UsbSerial in the Android Main.cpp ( https://github.com/XCSoar/XCSoar/blob/master/src/Android/Main.cpp ), and hopefully if this works make an PR for that.
Any objections, thoughts about this ?

iltis42, 2022-08-22:
Okay thanks for pointing out that. You say Android is the problem. Maybe, hence still have some questions:
Is the code you mentioned handling that under Android is the NativePortListener.cpp plus some generic files outside Android sources under Device/Port like TTYPort.cpp right ?
And do you have any idea why this utilities made from android-serialport-api, e.g. (1) obviously do get permission (2) and XCSoar not ?
1)
https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/android-serialport-api/SerialPort_1.1.apk
2) Video showing this ports in SerialPort apk: 
https://user-images.githubusercontent.com/34288016/185790549-fb3256bb-202b-4d29-9cbb-b0c60595c34f.mp4

MaxKellermann, 2022-08-22:
> Is the code you mentioned handling that under Android is the NativePortListener.cpp plus some generic files outside Android sources under Device/Port like TTYPort.cpp right

I fail to parse your question. If that helps: with native serial ports, NativePortListener is not used - that's only the bridge between Java-based port implementations and native code. Only TTYPort is used, and Java is not involved here.

> And do you have any idea why this utilities made from android-serialport-api, e.g. (1) obviously do get permission (2) and XCSoar not ?

I have no idea. Did you find any code which ensures that the permissions get granted?

MaxKellermann, 2022-08-22:
Ha, very funny:
https://github.com/xmaihh/Android-Serialport/blob/90320b3fddd67b46870a26d7e510870b805896e5/serialport/src/main/java/android_serialport_api/SerialPort.java#L209
```
                Process su;
                su = Runtime.getRuntime().exec("/system/bin/su");
                String cmd = "chmod 666 " + device.getAbsolutePath() + "\n"
                        + "exit\n";
                su.getOutputStream().write(cmd.getBytes());
```
XCSoar has the same ugly kludge:
https://github.com/XCSoar/XCSoar/blob/0251c31a0663bd6a35589b76cf0e0c3a05074b13/src/Device/Port/TTYPort.cpp#L199-L203
So even that, we already have.

Were you really able to use that other app on **your own Android device**, or did you just watch somebody else's video? Is your phone rooted, and is "su" installed?

iltis42, 2022-08-22:
My friend has this board on the table, i will get the same board this week, but i chat with him just now. The video was made on the rooted board he said and there is no "su" found in /system/bin. 
My next step is some debug logs to trace all devices coming in and the status if got filtered or not to get more light in here.

iltis42, 2022-08-25:
With a very small change (1) i can see now the desired devices (2) on my rooted board !!!
And i have an idea what went wrong: After some debugging i found that those device names _do_ appear in XCSoar, and _do_ have all permissions - rwx - well in place. See my logs (3) for the ttyS3, and (4) for the ttyS1 having - rw - only and that was the only one that appeared before (but wasn't usable either as dedicated to the manufacturers console). 

Now the situation is vice versa, the usable ports ttyS3,7,9 appear, and the one console port ttyS1 without executable flag doesn't.
So i have a good question:  Why we need to mask all ports with executable right's and those without executable rights not ?
The chmod above - 666 - ugly kludge for - rw - only may not work anymore from google restrictions or missing su or app permissions or whatever, so we see the permission as set in OS, what might be okay, maybe even better, or ?

Maybe we can also just ignore the executable flag ? Any objections ?

1)
<if (access(path, R_OK|W_OK) == 0 && access(path, X_OK) < 0)
>if (access(path, R_OK|W_OK) == 0 && access(path, X_OK) == 0)

2)
[image]

3) One of those ports of desire ttyS3,7 and 9:
2022-08-25T17:53:18Z] CheckTTYName ttyS7
[2022-08-25T17:53:18Z] CheckTTYName 4 ret=true
[2022-08-25T17:53:18Z] Next() access denied R_OK=0
[2022-08-25T17:53:18Z] Next() access denied W_OK=0
[2022-08-25T17:53:18Z] Next() access denied X_OK=0
4)  Unusable (manufacturer console) port:
[2022-08-25T18:02:34Z] CheckTTYName ttyS1
[2022-08-25T18:02:34Z] CheckTTYName 4 ret=true
[2022-08-25T18:02:34Z] Next() access (=0) okay   R_OK=0
[2022-08-25T18:02:34Z] Next() access (=0) okay   W_OK=0
[2022-08-25T18:02:34Z] Next() access (<0) okay   X_OK=-1

5) Lines that produced the above in TTYEnumerator.cpp/TTYEnumerator::Next()
 if (access(path, R_OK|W_OK) == 0 && access(path, X_OK) == 0){
LogFormat("Next() access (=0) okay   R_OK=%d", (int)access(path,R_OK));
LogFormat("Next() access (=0) okay   W_OK=%d", (int)access(path,W_OK));
LogFormat("Next() access (<0) okay   X_OK=%d", (int)access(path,X_OK));
:
}