XCSoar/XCSoar #1005: Kobo Clara HD not anymore supported , XCSoar won't load

varocketry, 2022-08-23:
XCSoar version
--------------

XCSoar 7.24 on Kobo

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
XCSoar should load and operate normally on the Kobo Clara HD platform.

XCSoar should load and operate normally on the Kobo Clara HD platform.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
After reading the recommendation by Tony and Nev at their paragliding website, I just purchased the Kobo Clara HD reader and have been attempting to load XCSoar software on it.

http://gethighstayhigh.co.uk/kobo-self-build/
The Kobo Clara HD is a high resolution, daylight readable e-reader than is well suited to use as a paragliding instrument.   I researched this unit and recommendation  fairly thoroughly but only just discovered that it is apparently not supported by newer software builds.

Previous ISSUES reporters have volunteered long term use on Kobo Clara HD unit for developers to perform their required testing.  
MAXKELLERMAN apparently released something that allowed him to CLOSE earlier reports, but the current software does not load on the Kobo Clara HD unit.

@GeekGoneOld
keith authored and [GeekGoneOld](https://github.com/XCSoar/XCSoar/commits?author=GeekGoneOld) committed on Dec 3, 2021
1 parent [9a3a111](https://github.com/XCSoar/XCSoar/commit/9a3a1112e75feca7a678f42daf13864aa75dab5d) commit 3d1a59d17ee707e67c37dff694ec9ed319fed520

MaxKellermann, 2022-08-23:
Other than "XCSoar won't load", you did not describe the nature of your problem. You have lots of useless blah-blah describing why the Clara HD is a great product, but that doesn't help here.

> MAXKELLERMAN apparently released something that allowed him to CLOSE earlier reports

What ..... ?????

MaxKellermann, 2022-08-23:
I just tested - XCSoar 7.24 works just fine on my Kobo Clara HD.

GeekGoneOld, 2022-08-24:
@varocketry I have been using Kobo Clara HD for about a year.  I agree with your research that it is an excellent platform for XCSoar.

I recently loaded 7.24 on it and it worked just fine.  I have also loaded a recent image (post 7.24) and it continues to work well.  Can you describe what is failing?

Note that I have also factory reset and reloaded the Clara HD on many occasions as I use it for XCSoar development.  It takes only a few minutes and I would suggest you try that then load 7.24.  Note you must save your Kobo stuff (books and settings).

MaxKellermann, 2022-08-24:
This bug report looks obviously bogus, and since it is also incomplete, I'm going to close it, instead of waiting for @varocketry to reply. If there is really a problem with Kobo Clara HD, write a better bug report.

closed 2022-08-24 by MaxKellermann

varocketry, 2022-08-25:
Max:

I was able to successfully load xcsoar latest version on the kobo Clara hd.

My problems were a combination  of poor instructions for a new user, and
user error.

I was trying to load the 8GB image onto a backup 32gb mictoSD card.  I was
initially enlarging the kobo partition to the available 22 Gb,  but it
failed to boot a dozen times. I thought to try without expanding the image
and it booted.

I also figured out that Koboroot.tgz file could not be renamed , in any
form, to be used. (Not mentioned  anywhere)

Thanks for your help,
Jim