XCSoar/XCSoar #1011: Error reading french airspace file with v7.25

fraktal-sb, 2022-08-25:
XCSoar versions having and not having the problem
-------------------------------------------------

problem detected with v7.25, does not exist on v7.23

Expected behavior
-----------------

the airspace file should be read without error message

Actual behavior
---------------

error message during airspace load

[image]

Steps to reproduce the behavior
-------------------------------

use frech airspace file from

https://www.ffvp.fr/liens-utiles-telechargements

and store it in XCSoarData/Airspace directory. Use this file in airspace setup

Do you have any idea what may have caused this?
-----------------------------------------------

software change (airspace data consitency checks)

bug or feature?

Do you have an idea how to solve the issue?
-------------------------------------------

no

lordfolken, 2022-08-25:
Feature. The file has an error.

lordfolken, 2022-08-25:
Previously it would ignore it and not display the particular airspace.

fraktal-sb, 2022-08-25:
so I will communicate this with the maintainer of the airspace file.

lordfolken, 2022-08-25:
Please use the one provided in the download manager. 
That one points to the github repo of the french airspace:
https://github.com/XCSoar/xcsoar-data-content/blob/master/data/remote/airspace/country/France_Airspace.txt.json

The project located here. 
https://github.com/planeur-net/airspace

As far as i can see that file is error free.

fraktal-sb, 2022-08-25:
but in https://xcsoar.org/download/data.html we have still the link to french airspace data

lordfolken, 2022-08-25:
Good catch. I removed the links. Soaringweb does the better job of maintaining this list. 
https://github.com/XCSoar/website/commit/c2cdf7b2f9554a95a013c7be79c5c29017fc7233

closed 2022-08-25 by lordfolken

Blaubart, 2022-09-03:
I have the same problem with the airspace from Poland

lordfolken, 2022-09-04:
Its hosted on soaringweb.org Want to take it up with john laibacher?

Blaubart, 2022-09-04:
OK, I'll send him a message

elgandoz, 2022-09-13:
Hey @lordfolken, we're getting this issue on the [Australia_HGPG_Siteguide.txt]().

The issues are all the powerlines crossing the valleys which are referenced using only 2 points:
```
AC Q
AN Powerline
AL SFC
AH 100 ft AGL
DP 36:41:33.64 S 146:54:35.88 E
DP 36:41:12.45 S 146:54:57.62 E
```

Given that this is valuable information, how can we draw a simple line?

MaxKellermann, 2022-09-13:
> Given that this is valuable information, how can we draw a simple line?

Second answer to this question: you can draw simple lines using XCSoar's topography renderer. You already see lots of lines on your map (roads, rivers etc.).

elgandoz, 2022-09-13:
@MaxKellermann  The topography renderer won't throw a warning on screen.

I get that a polygon is not a line, but the point is the safety aspect of flying, more than the semantic.

Winpilots OpenAir format defines the Q class for danger, which is what a powerline is.

Not having this type of warning on a free flying app would be a flaw IMHO.

MaxKellermann, 2022-09-13:
> Winpilots OpenAir format defines the Q class for danger, which is what a powerline is.

But it has no definition for "lines", therefore it is impossible to represent a powerline with a Winpilot OpenAir file.

> Not having this type of warning on a free flying app would be a flaw IMHO.

Then you should get involved in the "free flying app" and implement the feature. It's free software, you have the freedom to fix all flaws that you see, and you don't need anybody's permission or good will for that. Never complain about flaws. Just do it.

mlep, 2022-09-13:
> But it has no definition for "lines", therefore it is impossible to represent a powerline with a Winpilot OpenAir file.

Well, a Winpilot OpenAir file _can_ contain information about terrain: the file format ([http://www.winpilot.com/usersguide/userairspace.asp](http://www.winpilot.com/usersguide/userairspace.asp)) allows for this.

I think the exact issue is that this format separates the information into two files (airspace.txt and terrain.txt), but XCSoar does not support the terrain features at the moment.

MaxKellermann, 2022-09-13:
> I think the exact issue is that this format separates the information into two files (airspace.txt and terrain.txt), but XCSoar does not support the terrain features at the moment.

And the WinPilot file format still doesn't support lines; it only supports polygons, and a line is not a polygon.

elgandoz, 2022-09-13:
> Then you should get involved in the "free flying app" and implement the feature. It's free software, you have the freedom to fix all flaws that you see, and you don't need anybody's permission or good will for that. Never complain about flaws. Just do it.

I already did it in the past, and that's what I'm doing now.
I just found out, and I was just asking if there was a know way to have a line with the current implementation.
That's not a complaining.

mlep, 2022-09-13:
> it only supports polygons, and a line is not a polygon.

My understanding is that "open polygons" (ie. `TO {string}`) are to be used for roads, borders, etc.

MaxKellermann, 2022-09-13:
> My understanding is that "open polygons" (ie. `TO {string}`) are to be used for roads, borders, etc.

What exactly is an "open polygon"? Did they mean "polyline"? Because a polygon is never "open", by definition.

mlep, 2022-09-13:
Well, that may depend on your definition of a polygon. For example, this page, hosted by a math department, defines closed and opened polygons: [https://www.math.utah.edu/IBIS/man/node41.html](https://www.math.utah.edu/IBIS/man/node41.html)

Anyway, the format allows drawing roads, etc. The issue is about a suitable implementation in XCSoar.

MaxKellermann, 2022-09-13:
> Well, that may depend on your definition of a polygon. For example, this page, hosted by a math department, defines closed and opened polygons: https://www.math.utah.edu/IBIS/man/node41.html

This is documentation of some software, a copy of which just happens to be hosted by a math department of some university, but that does not mean this math department endorses the definition implied by this software manual.

Every math-related definition I could find on the net agrees with my description; i.e. there is no such thing as an "open polygon".

So let's not dilute the definition of "polygon", this only increases confusion, and distracts from the real problem, because we continue bikeshedding what a polygon is, when the (educated) rest of the world clearly agrees that a polygon must be closed.

The real problem is that the OpenAir format allows describing a **polyline** (which they mistakenly call "open polygon"), but XCSoar doesn't support that.

Also note that according to the OpenAir format specification, a polyline is declared with the `TO` command. The `AC` command, as discussed here, can *not* be used to describe a polyline, therefore no matter how you turn it, the data cited here is wrong.