XCSoar/XCSoar #1012: Native serial ports e.g. ttyS7, ttyS3 not seen under ANDROID target

iltis42, 2022-08-25:
XCSoar versions having and not having the problem
-------------------------------------------------
Android 11 and XCSoar 7.24 and older versions tested down to 7.20 don't see those interfaces

Expected behavior
-----------------
Open Device manager and all new device via native serial port e.g. ttyS1, etc 

Actual behavior
---------------
Those ports do not list up at all, similiar if under linux the user running XCSoar is not in the group dialout

Steps to reproduce the behavior
-------------------------------
Launch XCSoar on an android board with native serial devices and check device dialog which interfaces exist and which do not. Further info and testcase also here: #1002 

Do you have any idea what may have caused this?
-----------------------------------------------
Similar to linux when the user running XCSoar is not in the group 'dialout' where those interfaces aren't seen as well,
and commented by Max under #1002, "Android doesn't give XCSoar permissions to use those ports", its likely a permission problem. 

Do you have an idea how to solve the issue?
-------------------------------------------
Not really, any idea's how to workaround, or fix this would be great. 
Older app's on those boards linked with older SDKs still can (see #1002), so the issue IMHO comes with XCSoar in conjunction with newer SDK.  
A similar issue we have with scoped storage where new SDK linked apps won't be able to access folders in the files system. On OS level a serial /dev/ttyS3 device is nothing more than a file descriptor, hence also older XCSoar versions where exactly this works (7.20) also won't see the tty's so this might be also different story.
I have a board like this on my table, and can build for android debug version and do tests to get more light into this.

iltis42, 2022-08-26:
Now i have found a way to get the serial ports in service. The reason why they are masked is that on my board running Android 11, lays in the executable rights on their those ports have on their descriptor (1), while XCSoar ignores those ports at:
https://github.com/XCSoar/XCSoar/blob/c81dd1c802fd1b6479e7145fe167530b75d6b068/src/Device/Port/TTYEnumerator.cpp#L69
TTYPort.cpp attempt's to give the XCSoar process permissions to access the character device files by the below snippet, hence with recent Android version on new boards this won't work, leaving the executable flag as is on the device file.
`StringFormat(command, MAX_PATH, "su -c 'chmod 666 %s'", path);`

Further tests showed that executable rights are not affecting anything regarding the serial interface. With or without executable rights we can set baud rate, receive or sent data, the devices are running all properly. 
The conclusion of this:
There is no use at all for executable rights here. Kernels allow to set this flag, but won't allow to execute device files (2) either,  so this flag is mostly useless at all and for our application and can be ignored. The corresponding pull request will remove the check  executable rights must not exist and allows devices with those rights either.
The fix will unblock native serial ports running under android with executable flag set by removing this check. No negative side effect expected but a big benefit to use new hardware running newer Android versions.

1) Serial port with executable rights:
2022-08-25T17:53:18Z] CheckTTYName ttyS7
[2022-08-25T17:53:18Z] Next() access denied R_OK=0
[2022-08-25T17:53:18Z] Next() access denied W_OK=0
[2022-08-25T17:53:18Z] Next() access denied X_OK=0

Lines that produced the above in TTYEnumerator.cpp/TTYEnumerator::Next()
LogFormat("Next() access (=0) okay R_OK=%d", (int)access(path,R_OK));
LogFormat("Next() access (=0) okay W_OK=%d", (int)access(path,W_OK));
LogFormat("Next() access (<0) okay X_OK=%d", (int)access(path,X_OK));

2)
root@hp:/dev# ls -al ttyS31
crw-rw---- 1 root dialout 4, 95 Aug 20 12:23 ttyS31
root@hp:/dev# chmod +x ttyS31
root@hp:/dev# ls -al ttyS31
crwxrwx--x 1 root dialout 4, 95 Aug 20 12:23 ttyS31
root@hp:/dev# ./ttyS31 
-bash: ./ttyS31: Permission denied

Test that showed deviced operational on this android board and further info also documented here: https://github.com/XCSoar/XCSoar/issues/1002#issuecomment-1227641155

lyusupov, 2022-08-26:
For me it looks like 100% issue of the particular Android / Linux build or installation procedure but not the XCSoar.

Character device files in the /dev directory are to be created when Linux serial port driver proceed through 'device attachment' stage - this typically happens at the device boot time. 'Mknod' utility or system call is in use to map the device node into the file system space. 'Executable' flag is one of the arguments for the 'mknod' system call (or utility).

**Under regular circumstances, the 'executable' flag is not set when 'mknod' operations takes place.**

If would have such an issue on a Linux based system - I would inspect first if a particular UDEV system rules (or equivalent) for the serial device driver are defined properly. Unwanted hidden 'executable' flag may appear in the device driver configuration space.

lordfolken, 2022-08-27:
Chmod 666 removes the execute bit. 777 would be execute for user group and others (dangerous and bad practice) correct way woukd be to make the executing user part of a group that has read write permissions to the device file. 

The two places in the code are in contradiction. Best is to remove the execute bit, as only rw is needed for serial to work.

iltis42, 2022-08-27:
The following pull removes the check for executable bit:
https://github.com/XCSoar/XCSoar/pull/1018/commits/c8b07b48350b50bd6e7ac44a272d16a211a3cbb5

closed 2022-09-01 by MaxKellermann