XCSoar/XCSoar #1014: Crashes on Kobo on map pan

twhitehead, 2022-08-25:
XCSoar versions having and not having the problem
-------------------------------------------------

Version that crashes: 7.25
Version that didn't crash: 6.8.6 (haven't tried any versions in between)

Expected behavior
-----------------

I should be able to pan the map under all circumstances.

Actual behavior
---------------

If I wait for the map to load, and then I pan the map without first zooming way out or way in, it crashes. By crashes I mean that immediately upon entering pan mode I see a frozen pan screen with the top set of info boxes greyed out (not yet filled in with map), after a few moments the Kobo menu that lets you start XCSoar comes up, and there is a core file left in the _crash_ directory.

It does not crash if I zoom in or out significantly before panning, or if I pan immediately upon entering before it has had a chance to fully show the map on the screen. After I pan successfully once in a session (though any of these three ways), then I seem to be able pan further in that session without issues.

It doesn't matter if I bring up the menu first or not. It also doesn't matter if I start the pan with the pan gesture or the pan button in the nav menu.

Steps to reproduce the behavior
-------------------------------

Presumably if you installed it on a Kobo, used [my map file](https://github.com/twhitehead/soaring-ontariomaps) ([direct link to xcm file](https://staff.sharcnet.ca/tyson/ontario.xcm)), and started with a mid-level zoom centered on the London Soaring Society/Club, you would see the same thing?

Do you have any idea what may have caused this?
-----------------------------------------------

I feel that it crashes with the top info boxes greyed out and not yet filled in with map interesting. Perhaps there is some sort of initialization or race issue in whatever is done to extend the map display to the top of the screen in pan mode?

Do you have an idea how to solve the issue?
-------------------------------------------

I have copied the [core file from my kobo](https://staff.sharcnet.ca/tyson/core.1009.tar.gz) and I am hoping you can backtrace it, at which point what is going on might be obvious to you.

If there is anything at all that would help you track this down, please do not hesitate to ask. I am have spent quite a bit of time messing around with XCSoar and the kobo as should be evident from my map work and [kobo hacking notes](https://github.com/twhitehead/notes-kobohacking).

MaxKellermann, 2022-08-26:
This looks like a data race between the DrawThread and the main thread.

twhitehead, 2022-08-26:
Did a manual bisect with all version that I could download from the [releases directories](https://download.xcsoar.org/releases/) to narrow down the range.

Result was

* last good: 6.8.17
* first bad: 7.4

I couldn't narrow it down further as the server had no releases between these two.

MaxKellermann, 2022-08-29:
You did something wrong with your bisect. Bisect works on commits, not on releases.

twhitehead, 2022-08-29:
By manual bisect I meant I used the algorithm (a manual binary search) acrross the downloadable releases). I didn't use the git command. I would do that too, but it will requires actually getting it building.

twhitehead, 2022-08-30:
Built and tested the master branch (c81dd1c802fd1b6479e7145fe167530b75d6b068). It also crashes. Here are the details
```
(gdb) file output/KOBO/bin/xcsoar-ns 
Reading symbols from output/KOBO/bin/xcsoar-ns...
(gdb) target core output/KOBO/bin/core.640 
warning: core file may not match specified executable file.
[New LWP 640]
[New LWP 641]
[New LWP 646]
[New LWP 647]
[New LWP 648]
[New LWP 649]
[New LWP 645]
Core was generated by `/opt/xcsoar/bin/xcsoar -fly'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  memcpy (dest=0x2aed82b1, src=<optimized out>, n=<optimized out>)
    at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/string/arm/../memcpy.c:89
89				x = *(u32 *)(s+3);
[Current thread is 1 (LWP 640)]
(gdb) info thread
  Id   Target Id         Frame 
* 1    LWP 640           memcpy (dest=0x2aed82b1, src=<optimized out>, n=<optimized out>) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/string/arm/../memcpy.c:89
  2    LWP 641           epoll_pwait (fd=9, ev=0x2ab4cbb8, cnt=16, to=-1, sigs=sigs@entry=0x0) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/linux/epoll.c:29
  3    LWP 646           __syscall () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/internal/arm/syscall.s:14
  4    LWP 647           __syscall () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/internal/arm/syscall.s:14
  5    LWP 648           __syscall () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/internal/arm/syscall.s:14
  6    LWP 649           __syscall () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/internal/arm/syscall.s:14
  7    LWP 645           Luminosity8::operator== (other=..., this=<synthetic pointer>) at ./src/ui/canvas/PortableColor.hpp:240
```

Here is a backtrace on the thread that crashed
```
(gdb) bt
#0  memcpy (dest=0x2aed82b1, src=<optimized out>, n=<optimized out>) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/string/arm/../memcpy.c:89
#1  0x00380ce6 in memmove (dest=dest@entry=0x2aed82b1, src=src@entry=0x2b287fba, n=n@entry=757) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/string/memmove.c:13
#2  0x000e7846 in std::__copy_move<false, true, std::random_access_iterator_tag>::__copy_m<Luminosity8> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:420
#3  std::__copy_move_a2<false, Luminosity8 const*, Luminosity8*> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:495
#4  std::__copy_move_a1<false, Luminosity8 const*, Luminosity8*> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:522
#5  std::__copy_move_a<false, Luminosity8 const*, Luminosity8*> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:529
#6  std::copy<Luminosity8 const*, Luminosity8*> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:620
#7  std::__copy_n<Luminosity8 const*, unsigned int, Luminosity8*> (__result=0x2aed82b1, __n=757, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algo.h:728
#8  std::copy_n<Luminosity8 const*, unsigned int, Luminosity8*> (__result=0x2aed82b1, __n=757, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algo.h:760
#9  GreyscalePixelTraits::CopyPixels (n=757, src=0x2b287fba, p=0x2aed82b1) at ./src/ui/canvas/memory/PixelTraits.hpp:210
#10 RasterCanvas<GreyscalePixelTraits>::CopyRectangle<PixelTraitsOperations<GreyscalePixelTraits>, GreyscalePixelTraits> (this=0x7eb234d8, this@entry=0x7eb234c8, x=<optimized out>, y=<optimized out>, w=757, h=272, src=0x2b287fba, src_pitch=758, src_pitch@entry=255, operations=operations@entry=...) at src/ui/canvas/memory/RasterCanvas.hpp:835
#11 0x000e5fde in RasterCanvas<GreyscalePixelTraits>::CopyRectangle (src_pitch=255, src=<optimized out>, h=<optimized out>, w=<optimized out>, y=<optimized out>, x=<optimized out>, this=0x7eb234c8) at src/ui/canvas/memory/RasterCanvas.hpp:840
#12 Canvas::Copy (this=this@entry=0x7eb236e0, dest_position=..., dest_size=..., src=..., src_position=...) at src/ui/canvas/memory/Canvas.cpp:386
#13 0x000e63a8 in Canvas::Stretch (this=this@entry=0x7eb236e0, dest_position=..., dest_size=..., src=..., src_position=..., src_size=...) at src/ui/canvas/memory/Canvas.cpp:479
#14 0x000a7d32 in Canvas::Stretch (this=this@entry=0x7eb236e0, dest_position=..., dest_size=..., src=..., src_position=..., src_size=...) at ./src/ui/canvas/memory/Canvas.hpp:353
#15 0x000a7c2e in MapWindow::OnPaint (this=this@entry=0x9e9000, canvas=...) at src/MapWindow/MapWindowEvents.cpp:151
#16 0x000aa6de in GlueMapWindow::OnPaint (this=0x9e9000, canvas=...) at src/MapWindow/GlueMapWindowEvents.cpp:394
#17 0x000e1df2 in WindowList::Paint (this=this@entry=0x931c08, canvas=...) at src/ui/window/custom/WList.cpp:235
[... cut ...]
#19 0x00381e0a in start (p=0x2b614d38) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/thread/pthread_create.c:150
#20 0x003875a4 in __clone () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/thread/arm/clone.s:30
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Here is a [link to a tarball](https://staff.sharcnet.ca/tyson/core.640_xcsoar-ns.tar.gz) containing the core file and the the non-stripped executable for you to investigate them further if you wish.

twhitehead, 2022-08-30:
Thanks for the fast turnaround. Unfortunately it is still crashing though. Looks like the second thread is position has now changed though. Here is a backtrace from the latest master (8c9e81c1ea94a63d2a6751fcf996a91225167d8a).
```
(gdb) (gdb) file output/KOBO/bin/xcsoar-ns
Reading symbols from output/KOBO/bin/xcsoar-ns...
(gdb) target core output/KOBO/bin/core.661
warning: core file may not match specified executable file.
[New LWP 661]
[New LWP 669]
[New LWP 670]
[New LWP 668]
[New LWP 667]
[New LWP 662]
[New LWP 666]
Core was generated by `/opt/xcsoar/bin/xcsoar -fly'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  memcpy (dest=0x2aed82b1, src=<optimized out>, n=<optimized out>) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/string/arm/../memcpy.c:89
89				x = *(u32 *)(s+3);
[Current thread is 1 (LWP 661)]
```
```
(gdb) info threads
  Id   Target Id         Frame 
* 1    LWP 661           memcpy (dest=0x2aed82b1, src=<optimized out>, n=<optimized out>) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/string/arm/../memcpy.c:89
  2    LWP 669           __syscall () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/internal/arm/syscall.s:14
  3    LWP 670           __syscall () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/internal/arm/syscall.s:14
  4    LWP 668           __syscall () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/internal/arm/syscall.s:14
  5    LWP 667           __syscall () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/internal/arm/syscall.s:14
  6    LWP 662           epoll_pwait (fd=9, ev=0x2ab4cbb8, cnt=16, to=-1, sigs=sigs@entry=0x0) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/linux/epoll.c:29
  7    LWP 666           __wake (cnt=1, priv=128, addr=<optimized out>) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/arch/arm/syscall_arch.h:40
```
```
(gdb) bt
#0  memcpy (dest=0x2aed82b1, src=<optimized out>, n=<optimized out>) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/string/arm/../memcpy.c:89
#1  0x00380ce6 in memmove (dest=dest@entry=0x2aed82b1, src=src@entry=0x2b287fba, n=n@entry=757) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/string/memmove.c:13
#2  0x000e7852 in std::__copy_move<false, true, std::random_access_iterator_tag>::__copy_m<Luminosity8> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:420
#3  std::__copy_move_a2<false, Luminosity8 const*, Luminosity8*> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:495
#4  std::__copy_move_a1<false, Luminosity8 const*, Luminosity8*> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:522
#5  std::__copy_move_a<false, Luminosity8 const*, Luminosity8*> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:529
#6  std::copy<Luminosity8 const*, Luminosity8*> (__result=0x2aed82b1, __last=0x2b2882af, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algobase.h:620
#7  std::__copy_n<Luminosity8 const*, unsigned int, Luminosity8*> (__result=0x2aed82b1, __n=757, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algo.h:728
#8  std::copy_n<Luminosity8 const*, unsigned int, Luminosity8*> (__result=0x2aed82b1, __n=757, __first=0x2b287fba) at /home/tyson/Soaring/Kobo/XCSoar/output/KOBO/lib/armv7a-a8neon-linux-musleabihf/include/c++/12.2.0/bits/stl_algo.h:760
#9  GreyscalePixelTraits::CopyPixels (n=757, src=0x2b287fba, p=0x2aed82b1) at ./src/ui/canvas/memory/PixelTraits.hpp:208
#10 RasterCanvas<GreyscalePixelTraits>::CopyRectangle<PixelTraitsOperations<GreyscalePixelTraits>, GreyscalePixelTraits> (this=0x7e8424d8, this@entry=0x7e8424c8, x=<optimized out>, y=<optimized out>, w=757, h=272, src=0x2b287fba, src_pitch=758, src_pitch@entry=255, operations=operations@entry=...) at src/ui/canvas/memory/RasterCanvas.hpp:835
#11 0x000e5fea in RasterCanvas<GreyscalePixelTraits>::CopyRectangle (src_pitch=255, src=<optimized out>, h=<optimized out>, w=<optimized out>, y=<optimized out>, x=<optimized out>, this=0x7e8424c8) at src/ui/canvas/memory/RasterCanvas.hpp:840
#12 Canvas::Copy (this=this@entry=0x7e8426e0, dest_position=..., dest_size=..., src=..., src_position=...) at src/ui/canvas/memory/Canvas.cpp:386
#13 0x000e63b4 in Canvas::Stretch (this=this@entry=0x7e8426e0, dest_position=..., dest_size=..., src=..., src_position=..., src_size=...) at src/ui/canvas/memory/Canvas.cpp:479
#14 0x000a7d2e in Canvas::Stretch (this=this@entry=0x7e8426e0, dest_position=..., dest_size=..., src=..., src_position=..., src_size=...) at ./src/ui/canvas/memory/Canvas.hpp:353
#15 0x000a7c1e in MapWindow::OnPaint (this=this@entry=0x9e9000, canvas=...) at src/MapWindow/MapWindowEvents.cpp:157
#16 0x000aa6d6 in GlueMapWindow::OnPaint (this=0x9e9000, canvas=...) at src/MapWindow/GlueMapWindowEvents.cpp:394
[... cut ...]
#10 0x0015e72c in Thread::ThreadProc (p=<optimized out>) at src/thread/Thread.cpp:138
#11 0x00381e0a in start (p=0x2b614d38) at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/thread/pthread_create.c:150
#12 0x003875a4 in __clone () at /home/tyson/Soaring/Kobo/XCSoar/output/src/musl-1.1.18/src/thread/arm/clone.s:30
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
You can find the new core file and non-stripped binary [here](https://staff.sharcnet.ca/tyson/core.661_xcsoar-ns.tar.gz).

MaxKellermann, 2022-09-01:
The data race was only the harmless tip of the iceberg. It looks like everything is wrong with DrawThread synchronization. How did XCSoar survive so long without anybody ever noticing this? D'oh!

closed 2022-09-02 by MaxKellermann

twhitehead, 2022-09-02:
Happy to report that I can no longer crash it. Thanks very much!  :+1: 

Also really liking the work that has been done in the 7 series on the pan interface and the analysis system.