XCSoar/XCSoar #1021: SYSTEM TIME AS 2003 01 11

LucasStange, 2022-08-27:
Xc soar using samsung s10e, after upgrade to 7.8 version appears a system time as 2003 01 11.
Im already try some actions 
Create a new profile
Desinstall and install again xc soar
TURN OFF battery economy, nfc, positioning... turn on again
Downgrade and Upgrade my android
Turn on adcance mode to get GPS time sync

Always same phenomenum happening
[image]

In sym mode is running well

I believe it is a bug in xc soar to collect time of s10e samsung after a samsung and xcsoar upgrade

lordfolken, 2022-08-27:
Download gps test in appstore and check the gps time. Potentially rollover issue with the chip.

lordfolken, 2022-08-27:
Also update agps data and install xcsoar v7.25

LucasStange, 2022-08-27:
> Assigned #1021 <https://github.com/XCSoar/XCSoar/issues/1021> to
> @LucasStange <https://github.com/LucasStange>.

Thanks, but seems just time can be change, not date...

lordfolken, 2022-08-27:
What date does gps test report?

LucasStange, 2022-08-27:
Any, just time. I can not find data there...
Do you know how too?

LucasStange, 2022-08-27:
Got it.
Actual date, seems a xc soar bug...

LucasStange, 2022-08-28:
https://groups.google.com/g/rec.aviation.soaring/c/aB__edTlPBU

Seems this rollover is a old issue... but not solved.

Best regards

Lucas Stange

lordfolken, 2022-08-28:
The problem is your phone. Its doesn't report the proper date. 

Some applications cheat by changing the date. 

XCSoar can't as you can upload the output to a OLC/WeGlide or other website, manipulation of the gps fixes would be considered cheating.

You need a new firmware on your phone (firmware, not operating system) or a new phone.

closed 2022-08-28 by lordfolken

LucasStange, 2022-08-28:
If it is my phone, how to explain GPS test showing right date and time?

LucasStange, 2022-08-28:
Solucionado o problema da data no xcsoar.
Samsung apoiou.

Falhas em Apps geralmente estão relacionadas a conflitos de versões, por
isso é necessário redefinir as preferencias dos apps para o padrão de
fábrica.
Por favor, siga os passos abaixo:
1- Redefinir preferencias: Configurações > Aplicativos > Menu mais (três
pontinhos no canto superior direito da tela) > Reinificar prefer. do app
2- Otimizar configurações : Configurações > Assistencia do aparelho e
Bateria > Otimizar agora.
3- Limpar os dados armazenados no App Google, essa função não irá causar na
perda de nenhuma informação sensível do consumidor, no caminho:
Configurações > Aplicativos > Google > Armazenamento > Gerenciar
Armazenamento > Limpar todos os dados

Não se preocupe, nenhuma informação sensível será perdida.