XCSoar/XCSoar #1023: set standby frequency of an airfield nearby

Blaubart, 2022-08-27:
if you choose one of the airfields nearby (Alternates), you often expect to have to land there. From my point of view, it would be practical if the frequency of this airfield would then be sent to the radio as a standby frequency. The function could also be activated or deactivated in the menu if someone thinks they don't want to use it.

DanD222, 2022-08-28:
Why not just use the “Set Standby Frequency” button in the Alternates list? That’s what I do when I anticipate landing somewhere.

I personally wouldn’t want anything changing my radio frequencies without my explicit direction, plus I can imagine a scenario where someone’s trying to dig through the menu trying to disable the feature for whatever reason when low and close to an airfield… overall, I’m not sure how beneficial this feature would be.

lordfolken, 2022-08-28:
There are devices like the air display that offer such a feature. (With a paid subscription to frequency data..)

Blaubart, 2022-08-28:
I know the Air Control Display. That‘s why I had the idea for this feature. 

I think you misunderstood my suggestion because of enable or disable the function. My idea was to enable it generally, if you think it is a good option or disable it generally if you think it is not a good option. I agree, it is not a good idea change this option if you are low

lordfolken, 2022-08-28:
There is a lot of things to consider. 
 - over regional air frequencies
 - distance to airport
 - distance to airspace encompassing airport
 - inability to set multiple frequencies in cup files
 - how to announce the change to the user
 - configuration / documentation etc.

MaxKellermann, 2022-09-01:
Since a button for this action already exists, what is this feature request really about? I don't quite understand.

Blaubart, 2022-09-01:
I'm not talking about pushing a button to set the frequency. I mean, I pick one of the nearest airports because I'm probably about to land there, and that sets the radio's standby frequency to that local frequency as well. This saves stress when landing at an unknown airfield.

@lordfolken: I don't understand why this is to consider. I choose an airfield out of the list of nearest airfields and the frequency of this airfield is sending as standby frequency if I have activated this option. The user can be informed about the setting of the frequency by a pop-up message, as it has been used for other things up to now

DanD222, 2022-09-01:
Where do you see the benefit of selecting an airfield if you’re about to land there? If I’m about to land somewhere, I already know at what altitude I’m going to arrive there from the map view, so all I do is just use the Set Standby Frequency button from the Alternates dialogue and that’s it.

Asking just in case I’m missing something..

lordfolken, 2022-09-01:
So you fly near an airport.. the frequency is preselected. All good. But then you need to listen to the atis.. what now? You try to set the frequency on the stdby, and the stdby is suddenly reset to the frequency of the airfield. 

You want to fly a final approach to an airfield in a competition. you need to report the 2minutes out. Suddenly new frequency as another airfield is closer.

Then you struggle to turn the option off.. on final glide.. at 250kph.

Your in controlled airspace. There is an airfield below.. suddenly new frequency by atc, the frequency you are trying to enter is overwritten. 

You have two frequencies selected.. and you are using dual watch to monitor the other frequency. Suddenly that frequency is changed without you knowing.

Today you have listening squawk zones, TMZs, RMZs etc. 

The only option i would consider is a popup that would allow you to say yes to the frequency change.  But even that logic would require quite a few things to consider. (altitude, proximity, current frequency, dual mode, etc.) 

But on my openvario with stick this is one long press on the lower right button, to open the alternates dialog, 3xright button and ok, and the frequency is selected on the active frequency of the radio. That's easy to commit to muscle memory and can even be done blindly.

Blaubart, 2022-09-01:
@DanD222: OK, maybe I missed something, I'll check it out on the plane.

@lordfolken: I think you completely misunderstand me. The frequency should not be **automatically** set to the next airport. That would be really stupid!! If I select one from the list of alternative airports, **then** its frequency should be set as a standby!

MaxKellermann, 2022-09-02:
If you select an airport by pressing the "goto" button, you can just as well press the "radio frequency" button. I don't think the "Goto" button should implicitly also set the radio frequency. Using both buttons may be commonly used, but combining them may confuse and annoy a big number of XCSoar users. I don't think that's a good idea, and I think it's acceptable to require the user to press two buttons for two actions.

closed 2022-09-02 by MaxKellermann