XCSoar/XCSoar #1025: Folders for waypoint details

abiweil, 2022-08-28:
I don’t see any text either, maybe it’s inside the section that´s commented out? All I see is the following text: “_XCSoar version
all_”

_Originally posted by @DanD222 in https://github.com/XCSoar/XCSoar/issues/1024#issuecomment-1229469776_

abiweil, 2022-08-28:
XCSoar Version all
The proposal is to arrange the waypoint-details (for example the picture-files of outlanding fields and airfields) separated from the main folder "XCSoardata". If you have hundreds of picture files, there is very difficult to find the usual datafiles, which are necessary for configuring XCSoar.
In the waypoint-details-file it should be possible to select a subfolder, in which the associated picture-files can be assembled.
If you fly in different regions of the world, it can be very useful to have subfolders for the picture-files. For example if you fly in the Alps, you want to have the collection of outlanding-fields in the subfolder "Alps" and the airfields of Namibia collected in the folder "Namibia".
This effects a much better overview over the used waypoint-details and it can save memory space by relocating the unused waypoint-details/subfolders in other memories like the SD-card.

lordfolken, 2022-08-28:
The problem with letting a user define a path inside a cup file, is that its very easily exploitable. (Path Traversal) which is why most applications dis recommend to allow a user to specify a path. 

What i would have wanted to do, is a container format like xcm for the waypoints. similar to cupx. That than contain images/pdf/text for waypoints.

abiweil, 2022-08-28:
For clarification: The file with the details is not a .cup-file, it is a simple .txt-file.
I never intended to change the .cup-waypoint-file. 
The content of the details-file looks like that:
[ABFALTERSBACH ALF C]#
image=Abfaltersbach.png

lordfolken, 2022-08-28:
I was talking about CupX not cup. Cupx are 2 concatenated zip files (!) one contains a cup file and the other contains a folder pics with pictures in it. Its supported by lxnav, soaringspot and landewiesen.at

the problem with the path doesn't change whether you use a cup file or a waypoints details file to specify the path to the picture.

abiweil, 2022-08-29:
Thank you for the explanation. In this case I need a description or a program to get my image-files in the cupx-file. Strepla is not able to do that, perhaps Seeyou. In the manual of XCSoar I can't find a hint.Actually it seemed to be the easiest way for me if XCSoar would accept a path in the .txt-file like[Hahnweide] #Image=details\hahnweide.png

lordfolken, 2022-08-29:
The cupx is overly complicated and difficult to handle. You almost need to seek inside the binary file for the 2nd zip signature, and then you need to decompress the file. With it the cache handling and other difficulties. 

i wrote a rough parser some time ago for cupx. using binwalk, a forensic tool. 
https://github.com/lordfolken/xcsoar-cupx-converter

XCSoar's XCM is a non-compressed zip file (-0). Its already downloadable and is seekable, so it wouldn't have to be decompressed. 

The structure i imagine:
Non-Compressed zip file: Maybe extension .xcd
 - cup waypoint file
 - waypoint_details file (optional)
 - pictures in a predefined sub folder pics/

* prexisting formats, which we have parsers for. 
* no special handling
* end user generate-able

abiweil, 2022-08-30:
That sounds good.

lordfolken, 2024-01-08:
@MaxKellermann would you be able to implement something along the way of my proposal above? 

I have tried, but hit my limit, with how to handle this.

lordfolken, 2024-01-08:
I would then add repository support and generate the xcd files in the repository from various sources.

Joerg-CH, 2024-10-03:
Hi abiweil,

a simple way to solve your (initial) proposal to arrange the pictures in different folders I use since long to store the Visual Operation Charts of aerodromes:

I created with a file manager a subfolder 'wpt_details' on the same level as the folders 'logs' or 'tasks' created by XCSoar itself.

On Windows64: XCSoarData\wpt_details
On Android: Android>media>org.xcsoar>wpt_details 
(or Android>data>org.xcsoar>wpt_details for XCSoar v7.42 and older)

In the 'wpt_details.txt', I wrote:

	[Mengen Hohentengen]
	image=wpt_details/EDTM.jpg
	file=wpt_details/EDTM-VOC.pdf
	...

Important is to use the slashes '/', also for Windows64. In fact, I have the same 'wpt_details.txt' and folders on both Android and Windows64.	

In the folder 'wpt_details', I just put the pictures as 'EDTM.jpg', or 'EDTM-VOC.pdf'

In the case of a JPG picture, it opens directly in XCSoar. The PDF will open in the standard PDF-viewer of your device, and can have more than one page.

This works well with Android, and with Windows64 (only JPG).

If you have different folders for Alps or Namibia, you just have to choose the corresponding 'wpt_details_Alps.txt' or 'wpt_details_Namibia.txt' in XCSoar-Config-System-site files.

abiweil, 2024-10-03:
Many thanks, I will try it!

From: Jörg Schuon 
Sent: Thursday, October 3, 2024 10:11 AM
To: XCSoar/XCSoar 
Cc: abiweil ; Author 
Subject: Re: [XCSoar/XCSoar] Archive file format for Waypoints, Waypoint Details and associated resources (images/pdfs) (Issue #1025)

Hi abiweil,

a simple way to solve your (initial) proposal to arrange the pictures in different folders I use since long to store the Visual Operation Charts of aerodromes:

I created with a file manager a subfolder 'wpt_details' on the same level as the folders 'logs' or 'tasks' created by XCSoar itself.

On Windows64: XCSoarData\wpt_details
On Android: Android>media>org.xcsoar>wpt_details
(or Android>data>org.xcsoar>wpt_details for XCSoar v7.42 and older)

In the 'wpt_details.txt', I wrote:

[Mengen Hohentengen]
image=wpt_details/EDTM.jpg
file=wpt_details/EDTM-VOC.pdf
...
Important is to use the slashes '/', also for Windows64. In fact, I have the same 'wpt_details.txt' and folders on both Android and Windows64.

In the folder 'wpt_details', I just put the pictures as 'EDTM.jpg', or 'EDTM-VOC.pdf'

In the case of a JPG picture, it opens directly in XCSoar. The PDF will open in the standard PDF-viewer of your device, and can have more than one page.

This works well with Android, and with Windows64 (only JPG).

If you have different folders for Alps or Namibia, you just have to choose the corresponding 'wpt_details_Alps.txt' or 'wpt_details_Namibia.txt' in XCSoar-Config-System-site files.

abiweil, 2024-10-03:
Hi, Jörg,
many thanks, it seems to work. An additional question: Is the name of the wpt-folder arbitrary ( for example only alps, Namibia etc.) or has it to begin with wpt-details?
With regards

From: Jörg Schuon 
Sent: Thursday, October 3, 2024 10:11 AM
To: XCSoar/XCSoar 
Cc: abiweil ; Author 
Subject: Re: [XCSoar/XCSoar] Archive file format for Waypoints, Waypoint Details and associated resources (images/pdfs) (Issue #1025)

Hi abiweil,

a simple way to solve your (initial) proposal to arrange the pictures in different folders I use since long to store the Visual Operation Charts of aerodromes:

I created with a file manager a subfolder 'wpt_details' on the same level as the folders 'logs' or 'tasks' created by XCSoar itself.

On Windows64: XCSoarData\wpt_details
On Android: Android>media>org.xcsoar>wpt_details
(or Android>data>org.xcsoar>wpt_details for XCSoar v7.42 and older)

In the 'wpt_details.txt', I wrote:

[Mengen Hohentengen]
image=wpt_details/EDTM.jpg
file=wpt_details/EDTM-VOC.pdf
...
Important is to use the slashes '/', also for Windows64. In fact, I have the same 'wpt_details.txt' and folders on both Android and Windows64.

In the folder 'wpt_details', I just put the pictures as 'EDTM.jpg', or 'EDTM-VOC.pdf'

In the case of a JPG picture, it opens directly in XCSoar. The PDF will open in the standard PDF-viewer of your device, and can have more than one page.

This works well with Android, and with Windows64 (only JPG).

If you have different folders for Alps or Namibia, you just have to choose the corresponding 'wpt_details_Alps.txt' or 'wpt_details_Namibia.txt' in XCSoar-Config-System-site files.

Joerg-CH, 2024-10-03:
Hello abiweil,

When I first read about this wpt_details.txt, I thought it needs to have this name.
So I began with that, and kept it (I have currently only one file). But it should work also with files like , Alps.txt, Namibia.txt, also with folders containing the respective JPGs named Alps, Namibia.
I did not test subfolders, but this could also work.
The path in the details.txt needs only to be the correct one. The main thing is to use in the details.txt file always the slash '/', even when in windows the path is xcsoar\Alps.

Jörg

abiweil, 2024-10-04:
Just for Info! I have a Poke2Colorwith Android 9, there has the wpt_details_folder to be stored in the XCSoardata-directory! 
/storage/emulated/0/XCSoardata is indicated for the “Standortdatei”.
From: Jörg Schuon 
Sent: Friday, October 4, 2024 1:03 AM
To: XCSoar/XCSoar 
Cc: abiweil ; Author 
Subject: Re: [XCSoar/XCSoar] Archive file format for Waypoints, Waypoint Details and associated resources (images/pdfs) (Issue #1025)

Hello abiweil, 

When I first read about this wpt_details.txt, I thought it needs to have this name. 
So I began with that, and kept it (I have currently only one file). But it should work also with files like , Alps.txt, Namibia.txt, also with folders containing the respective JPGs named Alps, Namibia. 
I did not test subfolders, but this could also work. 
The path in the details.txt needs only to be the correct one. The main thing is to use in the details.txt file always the slash '/', even when in windows the path is xcsoar\Alps. 

Jörg

Joerg-CH, 2024-10-04:
Then your path is similar to the windows version, there it is also the XCSoarData folder, under which all files and the wpt_details folder is located.
My Android is 11, there I have the path /storage/emulated/0/Android/media/org.xcsoar shown in 'Site Files' /  'Standortdateien'. But it is sufficient to have the wpt_details.txt file and the wpt_detais folder in the same directory to keep the links working.
I even use the same wpt_details.txt both in Android for flying and in Win64 for testing / finetuning the data.
Be sure to have exactly the same waypoint names in your waypoint file and in the wpt_details.txt, so XCSoar can find the right data. I had first some problems with waypoint names like St. Marie, which is spelled in some wpt-files St Marie, in others StMarie.
Jörg