XCSoar/XCSoar #1026: Icons in task administration no more readable on KOBO

fraktal-sb, 2022-08-28:
XCSoar versions having and not having the problem
-------------------------------------------------

OK 7.23
NOK 7.24

Expected behavior
-----------------

the icons in task administation are no more radable on KOBO (Glo HD, 300 PPI)

Actual behavior
---------------

on this screenshot of 7.24 icons are to small

[image]

on this screenshot of 7.23 icons are OK

[image]

Steps to reproduce the behavior
-------------------------------

check 7.24 against 7.23

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

no

MikeBravo15, 2022-08-29:
The same for Status icons

lordfolken, 2023-03-22:
@fraktal-sb can you retest with latest update?

fraktal-sb, 2023-03-23:
tested with current head on KOBO GLO (not HD)
icons still too small (see comparision 6.8.17/7.29)
[image]

ubx, 2023-03-23:
> @ubx since you have current master running. Do the icons now look good?

Screenshot from Libra 2
[image]

For my taste, still a little too small

MaxKellermann, 2024-02-27:
I bisected this and found the first bad commit: ff406bf664842298cf1c710cd9b99dd13b9e8298
Obviously, this is a bad change, because it hard-codes the output sizes of the icons at certain pixel sizes, completely ignoring the physical sizes.
@lordfolken!

MaxKellermann, 2024-02-27:
If it didn't, I'd have to revert the whole commit, because it had "side effects". That commit was obviously bad - not because it was buggy - but because this change was weird and undocumented.

I didn't follow those icon changes, but I remember I was wary of merging ff406bf664842298cf1c710cd9b99dd13b9e8298 due to this exact undocumented change, but I told myself "let's not complain again this time", because people tell me I'm complaining too much. But every time I decide to not complain, it comes back to me. That's why I prefer to be strict. I rather not have a feature than merge bad code that haunts me forever.

lordfolken, 2024-02-27:
Your change now would have broken all the iconography, if i hadn´t  merged f2d0e21 and ff406bf.

MaxKellermann, 2024-02-27:
What point are you trying to make? I don't get it.

lordfolken, 2024-02-27:
On Android these icons have a normal physical size, even though its a high dpi (600+dpi) display. 
On the kobo the icon is rendered at 1:1 resolution.

MaxKellermann, 2024-02-27:
No they don't. You misunderstood the code. XCSoar assumes the small icon is 96 dpi and the big icon is 192 dpi, and scales the bitmap for display accordingly. If you saw correct sizes on Android with your bad change, it's by chance, but not by design.

lordfolken, 2024-02-27:
This is also apparent when using software rendering (OPENGL=n GLES=n) and launch xcsoar with -dpi=300x300

[image]

vs native 96dpi
[image]

So the icon's in general are not scaled along with the dpi.

MaxKellermann, 2024-02-27:
True. The software renderer is only used on the Kobo where the factor just happens to be 1, so this is not a problem.
But if you use software rendering on hi-dpi (where it was never "officially" supported), it becomes obvious that code is still missing after all these years:
https://github.com/XCSoar/XCSoar/blob/0f0d6a63343c7f33e086f8086de0903a616c5ac5/src/ui/canvas/custom/ResourceBitmap.cpp#L39-L48
Nobody ever bothered to implement that, nobody cared because the only target that uses the software renderer is the Kobo.

But it has nothing to do with this bug.

lordfolken, 2024-02-28:
The kobo mini might have had 167DPI.
The Clara HD has 300dpi.

So the root cause is lack of scaling in relation to DPI.

Resizing all icons across all platforms doesn't address this issue.

MaxKellermann, 2024-02-28:
LoadStretch implemented on Kobo: 3e12462aee5562a9d3a59bda5bbf14a683fa8e1b

fraktal-sb, 2024-03-16:
OK, tested on KOBO GLO (not HD)

[image]
[image]

lordfolken, 2024-03-16:
Thank you for checking.

closed 2024-03-16 by lordfolken