XCSoar/XCSoar #1031: UsbSerial FTDI-Devices -- Missed characters on certain baudrates

henrik1g, 2022-08-30:
XCSoar versions having and not having the problem
-------------------------------------------------
v7.25 and v7.24 both affected. So this is unrelated to recent code changes in UsbSerialHelper

Expected behavior
-----------------
Reliably receive all characters send on the serial line.

Actual behavior
---------------
Depending on the baudrate some characters are missed. Which triggers other problems. See issue:

This is *NOT* a hardware/wiring issue since the UsbSerialTerminalApp 
https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal&hl=en_US&gl=US
can receive all characters reliably at all baudrates on the same hardware setup.
XCSoar usb serial support is based on
https://github.com/felHR85/UsbSerial
where as the working terminal app is based on 
https://github.com/mik3y/usb-serial-for-android
which seems to have more active recent development and at least judging from a quick look into how baudrates are handled on FTDI devices a more thorough implementation of the hardware characteristics.

Steps to reproduce the behavior
-------------------------------
Connect a FTDI usb-serial device to the USB-OTG of an android device.
Send NMEA to the device at different baudrates and watch the result in the above mentioned terminal app to confirm reliable setup.
Then watch the same NMEA stream as received by XCSoar in the device monitor.
NOT affected: 9600 19200 and 38400
MOST affected: 57600
LESS affected: 115200
I used this NMEA code sent each second.
```
$GPRMC,143618.00,A,2944.49207,N,09528.00179,W,0.235,357.14,300822,,,A*74
$PGRMZ,121,F,2*38
$GPGGA,143618.00,2944.49207,N,09528.00179,W,1,05,6.95,64.4,M,-23.7,M,,*5F
$PFLAU,0,1,1,1,0,,0,,*63

```
This was tested on FTDI4232 multiport device, but I think single port FTDI devices should be affected as well. Maybe someone who has one of those can test.
My single port Prolific PL2303 is NOT affected, but this adapter does not use the FTDI code of UsbSerial, so no wonder.

Do you have any idea what may have caused this?
-----------------------------------------------
Flawed low level hardware driver code in 
https://github.com/felHR85/UsbSerial

Do you have an idea how to solve the issue?
-------------------------------------------
switch to using 
https://github.com/mik3y/usb-serial-for-android
(or find the reason and hope for upstream fix of felHR85/UsbSerial)

henrik1g, 2022-08-30:
[image]
[image]

henrik1g, 2022-09-01:
From the logcat it seems that felhr.usbserial code can directly cause a crash of xcsoar. Not as a secondary effect through the corrupted NMEA as I thought previously.
```
[09-01 21:55:00.382 10368:10503 E/AndroidRuntime]
FATAL EXCEPTION: Thread-4
Process: org.xcsoar, PID: 10368
java.lang.ArrayIndexOutOfBoundsException: src.length=65 srcPos=2 dst.length=61 dstPos=0 length=62
	at java.lang.System.arraycopy(Native Method)
	at com.felhr.usbserial.FTDISerialDevice.copyData(FTDISerialDevice.java:527)
	at com.felhr.usbserial.FTDISerialDevice.adaptArray(FTDISerialDevice.java:508)
	at com.felhr.usbserial.UsbSerialDevice$WorkerThread.doRun(UsbSerialDevice.java:359)
	at com.felhr.usbserial.AbstractWorkerThread.run(AbstractWorkerThread.java:21)

[09-01 21:55:00.389 2855:9713 W/ActivityManager]
  Force finishing activity org.xcsoar/.XCSoar

```

henrik1g, 2022-09-01:
I will be leaving for vacation tomorrow, so I cannot follow up on this directly. 
When I am back I could try to switch to https://github.com/mik3y/usb-serial-for-android which would most probably solve both issues.
However I would want kind of a green light from the devs that such a change would be welcome or at least considered.
Also I have no knowledge at all when it comes to open source licenses. But https://github.com/mik3y/usb-serial-for-android/blob/master/LICENSE.txt sounds good to me...

Waiting for your input.

MaxKellermann, 2022-09-02:
This looks like your adapter was detected as FTDI, but does not actually talk the FTDI protocol (does not send FTDI status bytes), and so the UsbSerial library cuts away the first 2 bytes of each 64 byte chunk. That would also explain why #1032 triggers the UsbSerial library crash bug - 65 bytes violate the FTDI protocol, because the second chunk has an incomplete FTDI status header.

MaxKellermann, 2022-09-02:
It's actually just a different packet size; more than 64 bytes.

MaxKellermann, 2022-09-02:
btw. https://github.com/mik3y/usb-serial-for-android suffers from the very same bug.

closed 2022-09-02 by MaxKellermann