XCSoar/XCSoar #1039: Multiple USB devices with same VID:PID id nor working for CP2102

hjr, 2022-09-05:
XCSoar versions having and not having the problem
-------------------------------------------------
7.24, 7.25, 7.26 on Android

Expected behavior
-----------------
Plugged USB/serial converter show up in XCSoar the same way as in a USB diagnostics tool and are selectable as interface to devices .. even if they have the same VID:PID id.

Actual behavior
---------------
Does work fine for ie. the FTDI uarts, but not for CP2102 (10C4:EA60) ones.

Steps to reproduce the behavior
-------------------------------
Plug multiple USB/serial converter and see the listed devices from XCSoar compared to a USB diagnostics tool.

Do you have any idea what may have caused this?
-----------------------------------------------
Not really, the topic was expected to be resolved with issue #968.

Do you have an idea how to solve the issue?
-------------------------------------------
--

hjr, 2022-09-05:
Pics of XCS and USB diagnostics when four USB converter are connected to the Android. 
[image]
[image]

MaxKellermann, 2022-09-05:
Do CP2102 devices have a serial number? (I can't reproduce because I don't have one.)

lyusupov, 2022-09-05:
https://github.com/XCSoar/XCSoar/issues/966#issuecomment-1228363451

MaxKellermann, 2022-09-05:
Thanks, @lyusupov, this is what I suspected - not a XCSoar bug then, and nothing XCSoar can fix.

closed 2022-09-05 by MaxKellermann