XCSoar/XCSoar #1044: WIN64 target fails compilation in std::error:code to boost::json::error_code assignment in src/Math/Histogram

iltis42, 2022-09-10:
XCSoar versions having and not having the problem
-------------------------------------------------
commit 2286f868a322555f3bad7c843a8b87b6ca340221 (HEAD -> master, origin/master, origin/HEAD)
Author: Max Kellermann <max.kellermann@gmail.com>
Date:   Fri Sep 9 12:29:53 2022 +0200

Did also a recursive submodule init,

Expected behavior
-----------------
docker compilation  xcsoar-compile WIN64 should compile and link.

Actual behavior
---------------
docker compilation fails with target WIN64, all other targets are fine:

docker run --mount type=bind,source="$(pwd)",target=/opt/xcsoar -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest /bin/bash
root@26030651400f:/opt/xcsoar# xcsoar-compile WIN64
  AR      output/WIN64/dbg/util.a
  CXX     output/WIN64/dbg/src/json/Boost.o
.
.
.
CXX     output/WIN64/dbg/src/Math/Histogram.o
In file included from ./output/src/boost_1_80_0/boost/json/src.hpp:37,
                 from src/json/Boost.cxx:45:
./output/src/boost_1_80_0/boost/json/impl/parse.ipp: In function 'boost::json::value boost::json::parse(boost::json::string_view, std::error_code&, boost::json::storage_ptr, const boost::json::parse_options&)':
./output/src/boost_1_80_0/boost/json/impl/parse.ipp:46:10: error: no match for 'operator=' (operand types are 'std::error_code' and 'boost::json::error_code' {aka 'boost::system::error_code'})
   46 |     ec = jec;

Steps to reproduce the behavior
-------------------------------
Do you have any idea what may have caused this?
-----------------------------------------------
Have seen a new boost version had been untared:
 UNTAR   boost_1_80_0.tar.bz2 

Do you have an idea how to solve the issue?
As we see incompatible error codes are getting assigned, either this is no more supported by boost, or a wrong coding in the source.

lordfolken, 2022-09-12:
Happens in the ci too.

iltis42, 2022-09-12:
> Happens in the ci too.

Yup, docker build an CI

MaxKellermann, 2022-09-14:
This is because in Boost 1.80, `boost::system` insists that the conversion is only possible if the C++ standard library implements `std::mutex`. Sigh.

closed 2022-09-14 by MaxKellermann