XCSoar/XCSoar #1045: Unable to download flights from Flarm to Openvario/XCSoar

eku60, 2022-09-11:
Welcome! - We kindly ask that you:

  1. Fill out the issue template below - not doing so needs a good reason.
  2. Use the forum if you have a question rather than a bug or feature request.

The forum is at: https://forum.xcsoar.org

Thanks for understanding, and for contributing to the project!

XCSoar versions having and not having the problem
-------------------------------------------------

Which XCSoar version are you using when seeing the problem?  7.26
Which XCSoar version is the last one you know that did not show the problem?  7.15

Expected behavior
-----------------

Please describe what you expected to see/should happen.
When connecting to Flarm for flight download, a list to select the file to be downloaded should appear, the file should be selected and then be downloaded

Actual behavior
---------------

Please describe the exact symptoms that you were seeing instead.
When connecting to Flarm after some sconds an error message like this occurs: "Error Unable to load file list"
After that the display shows "no data" to/from Flarm and the flarm is blinking red, what always was so before, but then the error box comes on screen and you have to acknowlege.

Steps to reproduce the behavior
-------------------------------

The more time you spend describing an easy way to reproduce the behavior (if
this is possible), the easier it is for the project developers to fix it!
Start XCSoar with an connect Flarm, swith to NMEA-Settings, activate the Flarm connection, klick file download and wait...

Version 7.15 and maybe some following versions, then show a file list. I have this problem since 7.23 

Do you have any idea what may have caused this?
-----------------------------------------------

This might give us ideas we haven't thought of, feel free to suggest causes.
Software changes in connections via ttys, Timeout Problems, connection problems.
Used: Flarm redbox on ttys1 with 38400 Baud, working nice during flight, that is when no write attempts are made.

Do you have an idea how to solve the issue?
-------------------------------------------

This might give us ideas we haven't thought of, feel free to suggest solutions.
Something must have been changed when establishing writing connetions via ttys

iltis42, 2022-09-11:
If you are testing with Android (can you comment if this is the case ?), this issue is a duplicate of #992, i have downgraded to 7.23 as of this issue also on my android tablet.

eku60, 2022-09-11:
Hallo,

 

it is Openvario, so no Android, but Debian Linux.

 

Kind Regards

eku60

MaxKellermann, 2022-09-23:
I don't have a FLARM so I can't help.

eku60, 2022-10-02:
Hello,

has anyone a Flarm or a LX Red Box in spare to lend it to Max Kellermann to solve this problem?
I would have to retract one from a plane (not so easy), but will not be able to do so in Oktober/Novmber because they are used for flying.

I think this is a problem more or less many users have, isn't it?
Help would be very mauch appreciated.

Regards
eku60

DanD222, 2022-10-02:
I can lend mine. I’ll be able to collect it from the glider next Sunday when I get to the airport.

eku60, 2023-05-15:
Issue closed, because the problem is solved since/in version 7.29. Tested with several OVs.