XCSoar/XCSoar #1047: XCSOAR 7.24 Kobo - Device A Port ttymcx0 missing

varocketry, 2022-09-12:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

 Under FLY|Menu|Config|Devices|Port A. in the listing of available ports I expect to find  ttymcx0 and to be able to set baud rate to 115200, using BlueFly Vario.

BlueFlyVario website blog "In the devices menu of xcsoar  (Menu|Config|Devices) edit device A with the following settings:
Port: ttymxc0
Baud rate: 115200
Driver: Bluefly"

Actual behavior
---------------

Under FLY|Menu|Config|Devices|Port A. in the listing of available ports ttymcx0 is missing.
The A ports listed are ttya4 thru ttyef, then ttyx0 thru ttyzf.

Port ttymcx0 is specified port by vario developer, Alistair.

Steps to reproduce the behavior
-------------------------------

 Open XCSOAR on Kobo device, check under FLY|Menu|Config|Devices|Port A. whether the listing of available ports INCLUDES ttymcx0 .

Do you have any idea what may have caused this?
-----------------------------------------------

 None, problems generally reported with Kobo versions since 6.8 according to developer, Nev Almond, others.

Do you have an idea how to solve the issue?
-------------------------------------------

. Restore port ttymcx0 as an available port

lordfolken, 2022-09-12:
Can you please try v.26? Some changes where made in the device discovery.

lordfolken, 2022-09-20:
@varocketry can you try v7.26 please?

psa-avat, 2023-01-07:
I made a modification on my branch to enumerate only serial devices. 
It works on my side with my Kobo.

bool
File::IsCharDev(Path path) noexcept
{
//  struct stat st;
//  return stat(path.c_str(), &st) == 0 && S_ISCHR(st.st_mode);
  int fd = open (path.c_str(), O_RDWR | O_NONBLOCK) ;
  if (fd > 0){
    struct serial_struct serinfo ;
    int ret = ioctl (fd, TIOCGSERIAL, &serinfo) ;
    close(fd) ;
    if (ret >= 0) return true;
  }

  return false;
}

Should be tested on other platforms or built only for Kobos
Will propose a pull request soon

lordfolken, 2023-01-07:
please have a look at it with @cmoetzing there was already a discussion here: https://github.com/XCSoar/XCSoar/pull/891

lordfolken, 2023-01-07:
Without feedback i tend to close this bug report, as it was IMO addressed.

flyToby, 2023-02-08:
i have just installed Xcsoar on the new Kobo Libra 2 and have this same issue of the missing port. have there been any solutions

lordfolken, 2023-02-08:
I'm closing this, as people are not reading any of comments, not providing versions and in general 0 effort is put forward. This is a community project it isn't free its the effort you put in.

closed 2023-02-08 by lordfolken

psa-avat, 2023-02-08:
I've prepared a new version on my side which fix this issue.
I'll try to make a pull request next week when back home.
I've tested on my Clara2E but can share my installation gz file if you want
to test on other models before.
One other solution is to update the profile file manually.

Patrick

flyToby, 2023-02-08:
yes that would be great thanks

psa-avat, 2023-02-08:
I'll check this afternoon if I can get it remotely from my NAS
Give me your email and I'll send it to you

flyToby, 2023-02-09:
how do i send a private message

psa-avat, 2023-02-09:
My email is ***@***.***

flyToby, 2023-02-09:
i cant see that its just stars

psa-avat, 2023-02-09:
my email is now available on my profile page

varocketry, 2023-04-11:
Okay, now I need to find your profile page.

MaxKellermann, 2023-04-11:
@psa-avat, if you refuse to contribute to this project, please don't promote your unsupported fork here.