XCSoar/XCSoar #1051: RASP File Download save with wrong name convension

Developer66, 2022-09-16:
The RASP files are downloaded as e.g. "Germany (by Thermalmap.info).dat".
Before moving the download to the DownloadManager, it was xcsoar-rasp.dat.

If XCSoar get restarted the file can not be found again...
Maybe not bad at all, so i have always to download the newest data but quite not very well in all cases.

@lordfolken

lordfolken, 2022-09-16:
Agreed. I will have a look.

dspreitz, 2022-10-26:
Following manual work-around works:
Manually rename a downloaded RASP file (e.g. 'CzechRepublic (by Thermalmap.info).dat') to xcsoar-rasp.dat in the /Android/data/org.xcsoar/files/ folder.
After that you can go to Info --> Weather --> RASP and select a field and time.

roscosoarer, 2022-11-17:
The Canberra Gliding club in Australia routinely glides in wave to FL245. I note the RASP function but there is no RASP map for our region.  A RASP download for 200 m radius [or to the East] for 36S 148E called 'Australia-Cooma NSW'
[image]
 Soaring Area would be great.

lordfolken, 2022-11-17:
The rasp maps are rendered by volunteers around the world. We merely provide a link to them.
Maybe you can convince a rasp operator in Australia: http://www.ausrasp.com/
Instructions on how to generate the correct files:
http://www.drjack.info/twiki/bin/view/RASPop/XcSoar

The urls must be added here, from then on it will be available on all XCSoar versions greater than 7.27
https://github.com/XCSoar/xcsoar-data-content/tree/master/data/remote/rasp/region

si-gr, 2023-04-28:
Feel free to use my work https://github.com/XCSoar/XCSoar/pull/1180 https://github.com/XCSoar/XCSoar/pull/1175 in a new PR as I am seemingly not able to contribute to XCSoar.

closed 2023-04-28 by MaxKellermann