XCSoar/XCSoar #1053: Segmentation violation after map file download

bomilkar, 2022-09-18:
XCSoar versions having and not having the problem
-------------------------------------------------

The current pre-release of 7.27 shows the issue.
This bug is **platform dependent**. I've seen it on Raspbian (Pi4) and OpenVario (Cubie2). Not on Debian or Windows or Android.

Expected behavior
-----------------

Should not run into Segmentation Fault.

Actual behavior
---------------

To better observe what's going on I compiled with "SANITIZE=y DEBUG_GLIBCXX=y".
Running on Raspian bullseye on a Pi4:
make -j4 SANITIZE=y DEBUG_GLIBCXX=y TARGET_OUTPUT_DIR=output/ASAN
ulimit -c unlimited
export ASAN_OPTIONS=verify_asan_link_order=0
./output/ASAN/bin/xcsoar --simulator

When XCSoar is up: Setup->Config->SiteFiles->Map->Download select a map file.
When the download progress reaches the end it produces this output:

```
/usr/include/c++/10/debug/safe_iterator.h:328:
In function:
    __gnu_debug::_Safe_iterator<_Iterator, _Sequence, _Category>&
    __gnu_debug::_Safe_iterator<_Iterator, _Sequence,
    _Category>::operator++() [with _Iterator =
    std::__cxx1998::_List_iterator<Net::DownloadListener*>; _Sequence =
    std::__debug::list<Net::DownloadListener*>; _Category =
    std::forward_iterator_tag]

Error: attempt to increment a singular iterator.

Objects involved in the operation:
    iterator "this" @ 0x0xa6883ee0 {
      type = std::__cxx1998::_List_iterator<Net::DownloadListener*> (mutable iterator);
      state = singular;
      references sequence with type 'std::__debug::list<Net::DownloadListener*, std::allocator<Net::DownloadListener*>                  >' @ 0x0xa77d1b68
    }
Aborted (core dumped)
```

In gdb I see this:
gdb ./output/ASAN/bin/xcsoar --core=./core

The attached transcript from gdb has the following in line 317 ...
[bt full ASan.txt](https://github.com/XCSoar/XCSoar/files/9594245/bt.full.ASan.txt)

```
#5  DownloadManagerThread::OnCompletion(std::__exception_ptr::exception_ptr) (this=this@entry=0xa77d1b00, error=...) at src/net/http/DownloadManager.cpp:287
        listener = <optimized out>
        __for_range = Python Exception <class 'AttributeError'> 'NoneType' object has no attribute 'pointer':
empty std::__debug::list
        __for_begin = <error reading variable: Cannot access memory at address 0xd1>
        __for_end = <error reading variable: Cannot access memory at address 0x2e>
        __PRETTY_FUNCTION__ = "void DownloadManagerThread::OnCompletion(std::__exception_ptr::exception_ptr)"
        path_relative = {static SENTINEL = 0 '\000', value = {static SENTINEL = <optimized out>, value = 0x1922660 "src/Math/XYDataStore.cpp"}}

```
A simple backtrace makes it clear (in my mind) that it may break in :
```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0xb60cc364 in __GI_abort () at abort.c:79
#2  0xb634bcac in __gnu_debug::_Error_formatter::_M_error() const () at /lib/arm-linux-gnueabihf/libstdc++.so.6
#3  0x00ebdee0 in __gnu_debug::_Safe_iterator<std::__cxx1998::_List_iterator<Net::DownloadListener*>, std::__debug::list<Net::DownloadListener*, std::allocator<Net::DownloadListener*> >, std::forward_iterator_tag>::operator++() (this=<optimized out>) at /usr/include/c++/10/debug/safe_iterator.h:328
#4  __gnu_debug::_Safe_iterator<std::__cxx1998::_List_iterator<Net::DownloadListener*>, std::__debug::list<Net::DownloadListener*, std::allocator<Net::DownloadListener*> >, std::bidirectional_iterator_tag>::operator++()
    (this=<optimized out>) at /usr/include/c++/10/debug/safe_iterator.h:579
#5  DownloadManagerThread::OnCompletion(std::__exception_ptr::exception_ptr) (this=this@entry=0xa77d1b00, error=...) at src/net/http/DownloadManager.cpp:287
#6  0x00ec9170 in BindMethodDetail::WrapperGenerator<void (DownloadManagerThread::*)(std::__exception_ptr::exception_ptr) noexcept, &DownloadManagerThread::OnCompletion>::Invoke(void*, std::__exception_ptr::exception_ptr)
    (_instance=0xa77d1b00, args#0=...) at ./src/util/BindMethod.hxx:130
#7  0x00ef3e9c in BoundMethod<void (std::__exception_ptr::exception_ptr) noexcept>::operator()(std::__exception_ptr::exception_ptr) const (args#0=..., this=0xa77d1b1c) at ./src/util/BindMethod.hxx:78
#8  Co::InjectTask::OnCompletion(std::__exception_ptr::exception_ptr) (this=this@entry=0xa77d1b04, error=...) at src/co/InjectTask.cxx:91
#9  0x00ef46c8 in BindMethodDetail::WrapperGenerator<void (Co::InjectTask::*)(std::__exception_ptr::exception_ptr) noexcept, &Co::InjectTask::OnCompletion>::Invoke(void*, std::__exception_ptr::exception_ptr)
    (_instance=0xa77d1b04, args#0=...) at ./src/util/BindMethod.hxx:130
#10 0x00eb5a9c in BoundMethod<void (std::__exception_ptr::exception_ptr) noexcept>::operator()(std::__exception_ptr::exception_ptr) const (args#0=..., this=<optimized out>) at ./src/util/BindMethod.hxx:78
#11 Co::InvokeTask::promise_type::final_awaitable::await_suspend<Co::InvokeTask::promise_type>(std::__n4861::coroutine_handle<Co::InvokeTask::promise_type>) (coro=..., this=<optimized out>) at ./src/co/InvokeTask.hxx:72
#12 _ZL14DownloadToFileR10CurlGlobalPKc13AllocatedPathPNSt7__debug5arrayISt4byteLj32EEER16ProgressListener.actor(_ZL14DownloadToFileR10CurlGlobalPKc13AllocatedPathPNSt7__debug5arrayISt4byteLj32EEER16ProgressListener.frame *) (frame_ptr=<optimized out>) at src/net/http/DownloadManager.cpp:253
#13 0x00eec250 in Net::_ZN3Net16CoDownloadToFileER10CurlGlobalPKcS3_S3_4PathPNSt7__debug5arrayISt4byteLj32EEER16ProgressListener.actor(Net::_ZN3Net16CoDownloadToFileER10CurlGlobalPKcS3_S3_4PathPNSt7__debug5arrayISt4byteLj32EEER16ProgressListener.frame *) (frame_ptr=0xa857cc80) at src/net/http/CoDownloadToFile.cpp:39
#14 0x00ee6fc8 in std::__n4861::coroutine_handle<void>::resume() const (this=<optimized out>) at /usr/include/c++/10/coroutine:126
#15 Curl::CoRequest::OnEnd() (this=<optimized out>) at src/lib/curl/CoRequest.cxx:96
#16 0x00ef0358 in CurlResponseHandlerAdapter::FinishBody() (this=this@entry=0xa857cde4) at src/lib/curl/Adapter.cxx:84
#17 0x00ef0768 in CurlResponseHandlerAdapter::Done(CURLcode) (this=this@entry=0xa857cde4, result=result@entry=CURLE_OK) at src/lib/curl/Adapter.cxx:105
#18 0x00ee64f0 in CurlRequest::Done(CURLcode) (this=0xa857cde0, result=CURLE_OK) at src/lib/curl/Request.cxx:128
#19 0x00eef140 in CurlGlobal::ReadInfo() (this=0xb138a680) at src/lib/curl/Global.cxx:196
#20 BindMethodDetail::WrapperGenerator<void (CurlGlobal::*)() noexcept, &CurlGlobal::ReadInfo>::Invoke(void*) (_instance=0xb138a680) at ./src/util/BindMethod.hxx:130
#21 0x00ef7040 in BoundMethod<void () noexcept>::operator()() const (this=<optimized out>) at ./src/util/BindMethod.hxx:78
#22 DeferEvent::Run() (this=<optimized out>) at src/event/DeferEvent.hxx:75
#23 operator() (e=<optimized out>, __closure=0xa68845e0) at src/event/Loop.cxx:204
#24 IntrusiveList<DeferEvent, IntrusiveListBaseHookTraits<DeferEvent>, false>::pop_front_and_dispose<EventLoop::RunDeferred()::<lambda(DeferEvent*)> > (disposer=..., this=0xb1f1d0a0) at ./src/util/IntrusiveList.hxx:346
#25 EventLoop::RunDeferred() (this=this@entry=0xb1f1cc90) at src/event/Loop.cxx:203
#26 0x00ef8f24 in EventLoop::Run() (this=0xb1f1cc90) at src/event/Loop.cxx:310
#27 0x00f0fd18 in AsioThread::Run() (this=<optimized out>) at src/io/async/AsioThread.cpp:48
#28 0x01432f5c in Thread::ThreadProc(void*) (p=0xb1f1cc80) at src/thread/Thread.cpp:138
#29 0xb69ef4a8 in asan_thread_start(void*) (arg=0xbe8667f0) at ../../../../src/libsanitizer/asan/asan_interceptors.cpp:205
#30 0xb6433310 in start_thread (arg=0xa6884f80) at pthread_create.c:477
#31 0xb6190da8 in  () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
```

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Keeping in mind that I've only seen this in Raspbian and OV (similar platforms) it may well be, that this is a compiler or library bug.
Anyway: I don't understand what is going on, and what is supposed to happen in src/util/BindMethod.hxx .

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2022-09-23:
There is no segmentation violation in your bug report. This is an assertion failure, which is a completely different kind of crash.

closed 2022-09-23 by MaxKellermann

bomilkar, 2022-09-23:
I'm just working on it. It definitely does end with "Segmentation fault (core dumped)".

After several days of poking around I inserted a few printf() to compare what happens on Raspbian vs Debian. I've seen it only a couple of times on Debian. On Raspbian it depends on the file I'm downloading. There is one file which almost always produces a crash. It looks to me as if it's random issue, maybe corrupted memory. But I don't know how to debug this.

You closed it. However, here is a transcript of "bt full" in gdb after the Segmentation fault. I can continue to work on this if you like. Just re-open the Issue.

[run_pi4_2.log](https://github.com/XCSoar/XCSoar/files/9634273/run_pi4_2.log)

mlep, 2022-09-23:
@bomilkar It looks like it is closed with a fix. :wink:

bomilkar, 2022-09-23:
Thanks for pointing this out, @mlep !
Looking at these changes: never in my whole live would I have come up with this!
But now it's clear why it behaves different on different platforms: handling threads is definitely different timing on different platforms. But I was close: I noticed the loop in lines 287 and 288 of DownloadManager.cpp wasn't doing what the code said it should.
Thanks a lot to @MaxKellermann for fixing this ugly thing!

lordfolken, 2022-09-23:
i guess this was the same: https://github.com/Openvario/meta-openvario/issues/217