XCSoar/XCSoar #1054: Segmentation fault in Download WeGlide task when no task is declared or active

bomilkar, 2022-09-21:
XCSoar versions having and not having the problem
-------------------------------------------------

most recent pre-build of 7.27 on UNIX with new feature "WeGlide declared task download"

Expected behavior
-----------------

Should not crash with signal SIGSEGV when no task is declared. Instead it should continue after reporting the issue.

Actual behavior
---------------

Hitting the OK button on the Error widget with message "No Task" leads to a segmentation fault.

Steps to reproduce the behavior
-------------------------------

Nav->Manage->"Download WeGlide task"
If there is no (active) declared task, a widget "Error No task" will open.

[image]

Hitting the OK button will immediately lead to signal SIGSEGV with a core dump (if enabled).

Do you have any idea what may have caused this?
-----------------------------------------------

Looking into the core file with gdb shows this:

```
Core was generated by `./output/UNIX/bin/xcsoar -simulator'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  OrderedTask::Clone (this=0x0, tb=...) at /usr/local/include/c++/10.3.0/bits/unique_ptr.h:173
173	      pointer    _M_ptr() const { return std::get<0>(_M_t); }
[Current thread is 1 (Thread 0x7f3bcb47a380 (LWP 23290))]
(gdb) bt
#0  OrderedTask::Clone (this=0x0, tb=...) at /usr/local/include/c++/10.3.0/bits/unique_ptr.h:173
#1  0x00000000004ff6c8 in TaskActionsPanel::OnDownloadClicked (this=0x225b670)
    at /usr/local/include/c++/10.3.0/bits/unique_ptr.h:173
#2  0x00000000004fefde in operator() (__closure=<optimized out>) at src/Dialogs/Task/Manager/TaskActionsPanel.cpp:169
#3  std::__invoke_impl<void, TaskActionsPanel::Prepare(ContainerWindow&, const PixelRect&)::<lambda()>&> (__f=...)
    at /usr/local/include/c++/10.3.0/bits/invoke.h:60
#4  std::__invoke_r<void, TaskActionsPanel::Prepare(ContainerWindow&, const PixelRect&)::<lambda()>&> (__fn=...)
    at /usr/local/include/c++/10.3.0/bits/invoke.h:110
#5  std::_Function_handler<void(), TaskActionsPanel::Prepare(ContainerWindow&, const PixelRect&)::<lambda()> >::_M_invoke(const std::_Any_data &) (__functor=...) at /usr/local/include/c++/10.3.0/bits/std_function.h:291
#6  0x0000000000492e62 in std::function<void ()>::operator()() const (this=<optimized out>)
    at /usr/local/include/c++/10.3.0/bits/std_function.h:622
#7  0x00000000005ab6ae in Button::OnClicked (this=<optimized out>) at src/Form/Button.cpp:134
#8  0x00000000005ab512 in Button::Click (this=this@entry=0x2116570) at src/Form/Button.cpp:145
#9  0x00000000005ab56a in Button::OnMouseUp (this=0x2116570, p=...) at src/Form/Button.cpp:208
#10 0x00000000005d4abd in ContainerWindow::OnMouseUp (this=0x2273520, p=...) at src/ui/window/custom/ContainerWindow.cpp:170
#11 0x00000000005d4abd in ContainerWindow::OnMouseUp (this=0x7ffc0db53b38, p=...) at src/ui/window/custom/ContainerWindow.cpp:170
#12 0x00000000005d4abd in ContainerWindow::OnMouseUp (this=this@entry=0x7ffc0db53ad0, p=...)
    at src/ui/window/custom/ContainerWindow.cpp:170
#13 0x00000000005a9842 in WndForm::OnMouseUp (this=0x7ffc0db53ad0, p=...) at src/Form/Form.cpp:238
#14 0x00000000005d4abd in ContainerWindow::OnMouseUp (this=this@entry=0x13391e0, p=...)
    at src/ui/window/custom/ContainerWindow.cpp:170
#15 0x000000000048db00 in MainWindow::OnMouseUp (this=0x13391e0, p=...) at src/MainWindow.cpp:589
#16 0x00000000005d5d74 in UI::TopWindow::OnEvent (this=0x13391e0, event=...) at src/ui/window/poll/TopWindow.cpp:89
#17 0x00000000005d7c55 in UI::EventLoop::Dispatch (this=this@entry=0x7ffc0db53a00, event=...) at src/ui/event/poll/Loop.cpp:63
#18 0x00000000005aa377 in WndForm::ShowModal (this=this@entry=0x7ffc0db53ad0) at src/Form/Form.cpp:417
#19 0x00000000004a1db6 in WidgetDialog::ShowModal (this=this@entry=0x7ffc0db53ad0) at src/Dialogs/WidgetDialog.cpp:199
#20 0x00000000004fa32f in dlgTaskManagerShowModal () at src/Dialogs/Task/Manager/TaskManagerDialog.cpp:258
#21 0x000000000045cac7 in InputEvents::eventCalculator (misc=<optimized out>) at src/Input/InputEventsTask.cpp:109
#22 0x0000000000457f09 in InputEvents::processGo (eventid=<optimized out>, eventid@entry=57) at src/Input/InputEvents.cpp:467
#23 0x00000000004584ee in InputEvents::ProcessEvent (event_id=57) at src/Input/InputEvents.cpp:305
#24 0x0000000000454d6a in MenuBar::Button::OnClicked (this=<optimized out>) at src/Menu/MenuBar.cpp:85
#25 0x00000000005ab512 in Button::Click (this=this@entry=0x14941a0) at src/Form/Button.cpp:145
#26 0x00000000005ab56a in Button::OnMouseUp (this=0x14941a0, p=...) at src/Form/Button.cpp:208
#27 0x00000000005d4abd in ContainerWindow::OnMouseUp (this=this@entry=0x13391e0, p=...)
    at src/ui/window/custom/ContainerWindow.cpp:170
#28 0x000000000048db00 in MainWindow::OnMouseUp (this=0x13391e0, p=...) at src/MainWindow.cpp:589
#29 0x00000000005d5d74 in UI::TopWindow::OnEvent (this=0x13391e0, event=...) at src/ui/window/poll/TopWindow.cpp:89
#30 0x00000000005d7c55 in UI::EventLoop::Dispatch (this=this@entry=0x7ffc0db53e20, event=...) at src/ui/event/poll/Loop.cpp:63
#31 0x00000000005d51d6 in UI::TopWindow::RunEventLoop (this=0x13391e0) at src/ui/window/custom/TopWindow.cpp:215
#32 0x0000000000520cf4 in Main () at src/XCSoar.cpp:120
#33 0x0000000000520ea1 in main (argc=<optimized out>, argv=<optimized out>) at src/XCSoar.cpp:164
(gdb) 
```

With Address Sanitizer it reveals this:

```
xcsoar: src/Widget/EditRowFormWidget.cpp:402: unsigned int RowFormWidget::GetValueEnum(unsigned int) const: Assertion `df.GetType() == DataField::Type::ENUM' failed.
Aborted
```
 

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2022-09-21:
Your bug report contains two different bugs which require two different fixes. Please split the bug report.

bomilkar, 2022-09-21:
I'm sorry, I picked the wrong line.
The original bug report now has the correct message from Address Sanitizer.
@MaxKellermann : I hope this was your point.

closed 2022-09-21 by MaxKellermann

bomilkar, 2022-09-22:
Works as expected.
Thank you!