XCSoar/XCSoar #1062: Paragliding polars load wrong values

elgandoz, 2022-10-08:
XCSoar versions having and not having the problem
-------------------------------------------------
7.27 ❌ - 7.26 ✅

Expected behavior
-----------------
While creating a new "plane", I would like to load the polar for my paraglider type from the list with meaningful values.
(~30-60 Km/h speed range - reference weight ~100kg)

Actual behavior
---------------
The loaded polar for any paraglider type aircraft is way off the flight envelope, they seems referring to sailplane ones.

Steps to reproduce the behavior
-------------------------------
Load any "Para *" polar and check the values, in particular speeds and reference weight.

Do you have any idea what may have caused this?
-----------------------------------------------
There has been some code refactoring for the Polar/Store in the last release, it could probably be related to that.
NB: [the values in the PolarStore](https://github.com/XCSoar/XCSoar/blob/master/src/Polar/PolarStore.cpp#L208-L212) are legit and they are what it used to load

Do you have an idea how to solve the issue?
-------------------------------------------
Unfortunately it goes beyond by skills.

Screenshots
-------------------------------------------
| v7.26 | v7.27 |
|-|-|
|[image]|[image]|

MaxKellermann, 2022-10-17:
I can't reproduce the problem.

But your 7.26 screenshot cannot possibly be of version 7.26, because the "Dry Mass" row has been removed in 7.23 already (commit 51415b9181e943445d5f4e19fca7e5192cd916be). There's something very fishy with this bug report.

elgandoz, 2022-10-17:
Hey @MaxKellermann, you're right, the screenshot for 7.26 was the wrong one (I took screenshots for several versions and uploaded the wrong one) but the relevant one was the one for 7.27 with the bogus values for "Para EN-C/DHV2".

I uploaded new and correct screenshots.

I could reproduce this on 3 different Android devices without existing configuration (I just changed the units from Aus kts to Eu km/h) and I had it confirmed by another user.

These are the values that should appear:
```
  { _T("Para Competition"), 100, 0, 39.0, -1.0, 45.0,  -1.1, 64.0, -2.30,  0, 0.0, 0 },
  { _T("Para EN A/DHV1"), 100, 0, 29.0, -1.1,  34.0, -1.3, 44.0, -2.30, 0, 0.0, 0 },
  { _T("Para EN B/DHV12"), 100, 0, 29.5, -1.1, 37.0, -1.2, 50.0, -2.30, 0, 0.0, 0 },
  { _T("Para EN C/DHV2"), 110, 4.19, 33.0, -1.0, 39.0, -1.2, 56.0, -2.30, 0, 0.0, 0 },
  { _T("Para EN D/DHV23"), 100, 0, 33.0, -1.1, 41.0, -1.2, 58.0, -2.30,  0, 0.0, 0 },
  ```

closed 2022-10-28 by MaxKellermann