XCSoar/XCSoar #1068: Reach of the glider calculation is broken

bomilkar, 2022-10-17:
XCSoar versions having and not having the problem
-------------------------------------------------

This has been happening in pre-releases of 7.28 since September 26, 2022. The 7.27 release (and earlier) is still OK.

Expected behavior
-----------------

The "reach of glider" calculation marks the reachable area (given terrain, altitude, wind, reach mode, etc).
 A star-like shape is expected unless the terrain is really flat.

Actual behavior
---------------

The reachable area of the glider is always circular. It seems that the terrain information is not taken into account.

[image]

Steps to reproduce the behavior
-------------------------------

Turn on Reach_Mode in Config->System-Glide_Computer->Route
Then replay a flight log so that a reachable area should be displayed. Try 7.27 to see how it's supposed to look.

Do you have any idea what may have caused this?
-----------------------------------------------

It might be related to recent changes made to src/Computer/RouteComputer .

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2022-10-17:
Can't reproduce - I just made this screenshot with a577092dff2110827346ed927c2914e100515cb7
[image]

bomilkar, 2022-10-18:
After a "git pull" and starting from an empty PRF applying only minimal changes: Still the same symptom.
Can you share your PRF?
Here is mine:
[empty.zip](https://github.com/XCSoar/XCSoar/files/9808987/empty.zip)

MaxKellermann, 2022-10-28:
> This has been happening in pre-releases of 7.28 since September 26, 2022. The 7.27 release (and earlier) is still OK.

This is wrong. 7.27 is *NOT* ok.

closed 2022-10-28 by MaxKellermann