XCSoar/XCSoar #1070: Polar list

silverstate64, 2022-10-22:
XCSoar versions having and not having the problem
-------------------------------------------------

unknown - haven't had to create a new plane/polar in a while

Expected behavior
-----------------

be able to select an Arcus M polar

Actual behavior
---------------

no polar

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

fraktal-sb, 2022-10-22:
use "Config Plane New/Edit" and "Polar Para EN B/DHV12" and adapt it to your needs

fberst, 2022-10-22:
In the polar list there is already a polar for the Arcus, the Arcus M should be the same with a higher dry mass (which the user can easily adjust)

When i remember right there isn't any polar published by the manufacturer of these planes. If you have access to a distinct polar for the Arcus M please share it, than we can add it to the list. Otherwise there is nothing we can do about this.....

silverstate64, 2022-10-23:
Thanks for the responses.
I do have some Polar numbers for the Arcus M but i'm unsure of their origin. Below are the numbers for selecting the Arcus Polar in XCSoar and the Polar numbers I have. 
I've been using my numbers for 3 years.

Arcus select in XCSoar
Polar V     35kt              58kt              89kt
Polar W   -0.91kt          -1.3kt           -3.89kt

Reference Mass 761 lb

what i have:
Polar V       64kt             86kt             113kt
Polar W    -1.30kt         -2.06kt         -3.89kt

Reference Mass 1653 lb

fberst, 2022-10-23:
> Arcus select in XCSoar 
> Polar V 35kt 58kt 89kt >
> Polar W -0.91kt -1.3kt -3.89kt
> 
> Reference Mass 761 lb

That's obviously completely wrong....
The values in the xcsoar PolarStore.cpp:
` _T("Arcus"), 700, 185, 110, -0.64, 140, -0.88, 180, -1.47, 15.59, 50.0, 120, 430 }` (values in kg, km/h and m/s)

When i load it i get the exakt same wrong results as silverstate64. (only on android)
So it seems we have some bug when loading the values from the PolarStore.
And it seems we already have an reported issue for that: #1062

lordfolken, 2023-02-08:
So this has been imlemented in #1026

closed 2023-02-08 by lordfolken