XCSoar/XCSoar #1071: Android Emulated Directory Access

hdechelard, 2022-10-23:
XCSoar versions having and not having the problem
-------------------------------------------------

 XCSoar 7

Expected behavior
-----------------

 Find and copy files to XCSoarData directory

Actual behavior
---------------

 Emulated directory is not accessible by phone user

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------
[image]

MaxKellermann, 2022-10-24:
What is "Emulated directory"?
What does "XCSoar 7" mean? Is that the version having the problem or the version not having the problem? Please read properly what is asked from you before you submit a bug report.
I have no idea what your screenshot means and why you didn't post instructions how to reproduce your problem. I don't even know what your problem  is.

MaxKellermann, 2022-10-28:
No reply

closed 2022-10-28 by MaxKellermann