XCSoar/XCSoar #1074: BLE support in release 7.11 breaks compability with some devices

MikhailMalyutin, 2022-10-28:
There was vario device port FFVARIO with driver BlueFly Vario. All works fine, perfect connection and xcsoar see barometer. 

On newer release if i chose FFVARIO BLE i see error Unsupported bluetooth device.

MaxKellermann, 2022-10-29:
Can't reproduce your problem. I can connect to my BlueFly Vario just fine.

MaxKellermann, 2022-10-29:
Since you chose to ignore the questions asked in the bug reporting template, I choose to close this bug report. Sorry, support is only possible if you make it possible.

closed 2022-10-29 by MaxKellermann

MikhailMalyutin, 2022-10-29:
> Can't reproduce your problem. I can connect to my BlueFly Vario just fine.
This is difficult to reproduce. I have next device - http://freeflyvario.ru/. Looks like it support BlueFly driver.

MaxKellermann, 2022-10-29:
Oh, now I understand. You are not actually using a BlueFly vario, you only use the XCSoar's BlueFly vario driver with a product that is supposed to be compatible with it (which I had never heard of).

XCSoar does not support this "freeflyvario". Apparently it does something differently, in a way that needs explicit support from XCSoar. Maybe it announces itself as both BLE and "classic" Bluetooth, and XCSoar 7.11+ always prefers BLE, and because no "real" BlueFly vario can do BLE, that's not a problem with the original BlueFly.

To develop this, we need one such freeflyvario. Maybe you can convince the manufacturer to send us one accompanied with BLE protocol documentation, and then we might take the time to help them. 

(Please note that I won't support Russian products as long as your country keeps terrorizing and slaying innocent people in Ukraine. Let's talk again after Russia has retreated from foreign territory.)