XCSoar/XCSoar #1077: Failure of Auto-MC in Final-Glide-Mode above 180kph

Soarinator, 2022-11-14:
XCSoar versions having and not having the problem
-------------------------------------------------

 The version without this problem is eight or more years old

Expected behavior
-----------------

 In final-glide-mode I would expect a smooth transition of the values above 180kph.

Actual behavior
---------------

Auto-MC in final-glide-mode doesn't work @ speeds more than 180kph. MC is switching from latest value to MC 6,7 with fix speed 180kph when exceed it.

Steps to reproduce the behavior
-------------------------------

 You can reproduce the malfunction by using the simulation-mode

Do you have any idea what may have caused this?
-----------------------------------------------

 It's not a problem of polar-extrapolation. I made a test with a modified polar up to 250kph. I don't know the reason.

Do you have an idea how to solve the issue?
-------------------------------------------

 A version from 10 years ago works very well, there must be a change in between

Soarinator, 2022-11-14:
It's a mistake in translation to german. Max cruise speed was set to 180kph because it's "Manövergeschwindigkeit" in german what menans max. speed for using one rudder 100%.

closed 2022-11-14 by Soarinator

lordfolken, 2022-11-14:
That's actually a fundamental issue. The code assigns this to max_speed.
https://github.com/XCSoar/XCSoar/blob/e42f85410f17808b4f02bfd29d8bc012fe0a264b/src/Dialogs/Plane/PlaneDetailsDialog.cpp#L136-L138

I corrected the translation above.