XCSoar/XCSoar #1080: Sharing task information via QR code

Mitkopeev, 2022-11-22:
XCSoar version -7.28
--------------

7.28

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Sharing tasks via QR code

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

QR CODE can be used for sharing task information and to quickly load tasks directly from other users.
Currently there is no quick way for sharing task information within the app except direct task file transfer between devices.
Sharing task info via QR code is a quick, easy and network free way of sharing the information.
Extremely useful if pilot doesn't have network connection for example for task download directly from online servers or does not have waypoints present.

lordfolken, 2022-11-22:
Easiest way is to start to try to implement yourself.

lordfolken, 2023-02-20:
There is a library here that seems simple enough and MIT licensed.
https://www.nayuki.io/page/qr-code-generator-library#cpp

lordfolken, 2026-10-20:
Shipped in 7.46 by #2861.

The task manager shows the current task as a QR code, and on Android scans one back in. The code is the XCSoar task, so a line, a sector, a keyhole and the task rules are kept. When that payload does not fit, it falls back to a SeeYou CUP code, then to an XCTrack code if every zone is a cylinder.

closed 2026-10-20 by lordfolken