XCSoar/XCSoar #1083: Delay befor dissapearing of an FLARM-Info

erikkurtz, 2022-12-02:
XCSoar version
--------------

7.2.8 Android

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

A delay between last FLARM-Info and disappearing from the map or FLARM-Screen.
In best case I could adjust the delay in the configuration-menu.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

In LX (9070 and others) you can adjust the time how long you see the info after last signal. That's very useful, because even while thermaling often the FLARM-Infos have bad signal quality and often disappear.

groundhogxc, 2023-04-03:
Yes please!
Even better would be a second phase with a different icon which could be even longer (minutes),

And for a long time I have wished a similar function as what happens with Skylines "friends": Designating certain FLARMS as "friends" (for example by assigning one of the four colours as "friend"-color) should make their last known position permanently displayed (just like the skylines symbol, for example). That would be amazing.

closed 2023-05-24 by MaxKellermann

erikkurtz, 2023-05-24:
Thanks a lot! Looking forward to the next published version.

groundhogxc, 2023-06-01:
Looks very nice! The "faded" arrows are however a bit hard to see under not-super-perfect lighting/display conditions. It is fine when they have labels, though. Maybe the border of the arrow symbol could be a bit fatter? (both the "active" and the "faded" one)

Actually it seems it is right now a hair-line without any width, so not scaled with DPI - so it might be a problem only on devices with higher resolution.

MaxKellermann, 2023-06-01:
@groundhogxc check 0ad9d930a038a212b731891d75d7bbcd33a1bd0c