XCSoar/XCSoar #1084: Add support for Argentinean 95% distance rule

andrestotorica, 2022-12-08:
XCSoar version
--------------
7.28

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
[Argentinean federation contest rules](https://www.favav.com.ar/img/deportiva/Reglamento%20general%20agosto%202018.pdf) (item 3.7.2.5.c) have a **95% distance rule**. 

* In an AAT task, if a pilot flies more than 95% of the task maximum distance, then the computed task time is the actual flown time, even if lower than the task time.

In other words, if you fly close to the max distance and you are still early, you are not penalized, the actual task time is used.

Today, pilots have to calculate the 95% distance mentally, and have no cues whether the projected targets are within it.

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Feedback is welcome as to which helpers for the 95% rule would be more useful.

* Add option in settings (Configuration > Glide Computer > Route?) so users in Argentina can selectively enable the features.
* **AAT Max distance** infobox show the 95% of max distance in the secondary field
* **AAT distance around target** infobox show which percentaje of the max distance is projected, in the secondary field
* Maybe an arc on the map, showing the 95% of max distance in the last area? (needs more thinking)

I will get started on the first ones while ideas are being refined.

MaxKellermann, 2023-05-18:
Please upload task files and IGC files describing various outcomes (less than 95%, between 95% and 99%, 100%) - and say what the correct calculation results would be. That way, we can develop the feature and verify the correctness of the implementation.

MaxKellermann, 2023-06-29:
No reply

andrestotorica, 2023-07-05:
I somehow missed the last messages, sorry.
I planned to try an implementation as it was suggested by various pilots during a competition last summer, but ultimately I got sidetracked. I can retake working on this if there are no objections.

MaxKellermann, 2023-07-08:
There are never objections to people who want to improve XCSoar. Your work should contain a unit test so we can reproduce the results and can ensure it continues to work.

lordfolken, 2024-06-02:
@andrestotorica i see that you did work on this. Can you make a PR?

andrestotorica, 2024-06-03:
Hi @lordfolken, I did progress on the implementation but then found a deeper problem with AAT task distances that I couldn't solve, and eventually got stuck. I'll explain the issue below. If anyone more experienced can give me a hand with some guidance, I will try to get it completed.

___

### Issue
[This rather simple commit](https://github.com/andrestotorica/XCSoar/commit/1967ad8045728decf0652cb11cb07470b630fc53) is the hearth of the change on the infoboxes. However, it has a bug: the task's maximum distance is needed, but the value stored in `task_stats.distance_max` represents the max achievable distance from the present situation - i.e. it's a constantly updated value based on the track already flown.

This implies there's currently a minor UI bug. If you open the Task Manager screen mid-task, in particular after having passed the 1st area, the max value is going to be wrong. The _current achievable_ max is shown, instead of the _global_ task max. 
[image]
I can provide a more detailed repro if this is not clear, and open a separate issue if that would make sense. 

### Possible solutions
So, a new member is needed in `MapSettings` that stores the task's maximum distance at all times, to show in the Task Manager window and for the purposes of the new feature.
I can see two possible ways to address this:

1. Have the value be copied from the calculated max distance ONLY when the task is not started and modified in any way, and not updated as the flight progresses. Note this is quite different from how the min, max and nominal distances are calculated (which update continuously in `AbstractTask::UpdateStatsDistances`). 
Unfortunately I couldn't achieve this - maybe I'm misunderstading the `full_update` flag or it can't be done with the code as-is. This could also have implications if the task is modified after start, not sure if that is even a valid use case.

2. Have the task's maximum distance be treated similarly to how the nominal distance is calculated. This is, trigger a re-calc whenever, just have a separate method that calculates max distance over the whole task, ignoring already sampled points (it involves making a new `RunDijsktra...` method)
See this patch: [maxDistance.patch](https://github.com/user-attachments/files/15529332/maxDistance.patch). NOTE: this is not meant to be functional, or even correct, code. It's just an illustration of the solution proposed above. It would need closer inspection and refactoring.
To me, this approach sounds more solid that \# 1 as the value is just continuously calculated from the task, and doesn't depend on anything being preserved and not updated at an incorrect time. But it is computationally less efficient (I don't know if this can be even significant). 

I'd love to hear others opinions on whether this is worth pursuing, and see where we go from there.

andrestotorica, 2024-06-09:
Added PR with proposed change. This should be functionally correct and clean now.

closed 2024-12-17 by lordfolken