XCSoar/XCSoar #1085: Anything against surfacing FLARM data to Lua?

brunotag, 2022-12-09:
I'd be keen to code the change.

XCSoar version
--------------
Latest

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Surface FLARM data to Lua

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Enable customisations?

brunotag, 2022-12-10:
Thanks Max. 
Speaking of surfacing functionality to Lua, another thing possibly helpful could be to formalise those legacy events into lua tables?

E.g. 
```
xcsoar.fire_legacy_event("PlaySound", "scorreggia.wav")
```
could become
```
xcsoar.sound.play("scorreggia.wav")
```

and
```
xcsoar.fire_legacy_event("StatusMessage", "Wassup?")
```
could become
```
xcsoar.display.popup("Wassup")
```

and the undocumented
```
alert("modal text")
```
could become
```
xcsoar.display.modal("modal text")
```
?

Thoughts? 
And, if so, anything else that comes to mind?

I'd be keen to code this too, this would possibly already stretch my limited C++, but I'd be keen to, because afterwards I could get those goodies in Lua?

Maybe separate issues for each of the things?

MaxKellermann, 2022-12-10:
I'm cautious with wiring more legacy stuff into Lua, because that will create the expectation of a stable API that we must always keep compatible with. But 1. it's better to have something than to have nothing, and 2. the examples you named look reasonable.

When I decided to introduce Lua scripting in XCSoar, to replace the old and ugly InputEvent system, I envisioned that one day, people could write whole new XCSoar features in Lua, and only a small core would need to be kept in C++. That has not happened, mostly because I quitted flying, got 4 kids and therefore stopped putting another full-time job's worth of time into XCSoar (which I had been doing for 6 years or so). That vision is still something to look after, but these days, the time I have available for XCSoar is barely enough to keep the project running, fixing the most important bugs. That's bad, because Lua would enable many more people to contribute, so I'm happy you're here to help.

lordfolken, 2023-01-22:
I think the basic question has been answered. Therefore closing.

closed 2023-01-22 by lordfolken