XCSoar/XCSoar #1086: Integration of anemoi (fly-anemoi.de) data into XCSoar

Indiasieben, 2022-12-15:
XCSoar version
7.3

7.26

What should XCSoar do differently, what functionality should be added?
integration of anemoi wind data into XCSoar (fly-anemoi.de)----------------------------------------------------------------------

 anemoi is measuring the wind data with acceleration sensors. The output is shown in a separate display. It would be very useful to show the wind details mid-wind and current wind in XcSoar devices like OV.

What are you trying to do, what is the use case for the suggestion?
the measurement is very precise and helps to find the thermals faster like Hawk software in LX devices------------------------------------------------------------------

the I/O of the anemoi box is described

MaxKellermann, 2022-12-15:
On that website, I only see a white page with an animating circle.

Indiasieben, 2022-12-16:
Hi Max,

toll wäre es eine ähnlich präzise Windanzeige wie Hawk bei den LX-Geräten für die XCSoar-User zu haben. Das ist der Gedanke dahinter. Bei Stefly (Stefan Langer) ist eine sehr ähnliche Entwicklung im Gange. Vielleicht nimmst Du mit ihm Kontakt auf. Er kann es Dir sicherlich viel besser erklären als ich.

Vielen herzlichen Dank für Dein Interesse und all die tolle Arbeit, die Du im Laufe der Jahre für uns als XCSoar-User geleistet hast.

Ich hoffe, damit kommst Du weiter in Deinen Überlegungen.

Schönes Weihnachtstfest und guten Rutsch ins Neue Jahr.

Always happy Landings.

Bernd

 

Gesendet: Donnerstag, 15. Dezember 2022 um 22:42 Uhr

dragonfly771, 2022-12-19:
I think we should split this topic in two parts: 

- new infobox for dual wind or a separate infobox for the second wind
- driver to communicate with external device which provides the wind data (anemoi; Larus Breeze)

For the Larus project we suggest to extend the OpenVario driver with the addition of AHRS and real time wind sentence as following: https://github.com/Witloofboer/openvario-protocol 

More information about Larus Breeze on Github: https://github.com/larus-breeze/sw_sensor_algorithms

Indiasieben, 2022-12-19:
very good idea!

 
 

Gesendet: Montag, 19. Dezember 2022 um 09:27 Uhr

August2111, 2023-01-30:
Hi Bernd,
since last week I work an this topic. On my desk it works very well. At the end of the day I would have the driver for both: Anemoi and also Larus too. In the moment a big question is, how I can insert my investigations in XCSoar itself, but hopefully we have a solution for this up to end of February. So the (European) flight season  can start with this :)

Indiasieben, 2023-01-30:
. Great news. Thanks a lot for your efforts. 

 
 

Gesendet: Montag, 30. Januar 2023 um 15:36 Uhr

August2111, 2023-02-06:
Hallo Bernd,

wegen der Integration der Treiber würde ich gerne einmal direkt mit Dir 
Konakt aufnehmen wollen :)

Kannst Du Dich einmal kurz unter dieser meiner eMail-Adresse melden?

Gruß

August

August2111, 2023-02-06:
Hi Bernd, I have very good contact with Stefan, I am also connected to him through the Larus group :)
But please write to me directly at <Info@FlapsOnline.de> - in this international forum no German text is actually desired (especially not with attachments)
Regards
August

Indiasieben, 2023-02-06:
I will do so! Thanks a lot for the hint, sorry for that.
Regards 
Bernd

MaxKellermann, 2023-05-16:
I tried hard, but couldn't find any technical documentation of the protocol which would allow XCSoar developers to write a driver. Therefore, this product cannot be supported by XCSoar currently.

closed 2023-05-16 by MaxKellermann

Indiasieben, 2023-05-16:
Hi Max,

thanks a lot for the information. I'll try to contact Benny Bachmaier, and I’ll ask him to contact you for sharing the needed informations.

Best regards

Bernd

 
 

Gesendet: Dienstag, 16. Mai 2023 um 15:51 Uhr

MaxKellermann, 2023-05-16:
Nobody needs to contact me, just making the documentation available to me isn't enough, this doesn't ensure all pilot's rights to use the hardware they paid for.
The protocol documentation should be available for free for anybody, without registration and without personal access. Once that happens, we can reopen this feature request.