XCSoar/XCSoar #1087: GSP Week Number

PascalLauper, 2022-12-21:
XCSoar versions having and not having the problem
-------------------------------------------------

V7.28-Android

Expected behavior
-----------------

When selecting Info-Page 2/3-Etat-Hours, I am expecting that the Date UTC is correct, up to date

Actual behavior
---------------

on 2022-12-21 the UTC date reported is 2003-05-07. 

Steps to reproduce the behavior
-------------------------------

 it is always present with XCSoar V7.28-Android application, which I am running on a SAMSUNG S9+ Android Version 9.

Do you have any idea what may have caused this?
-----------------------------------------------

I am the believe that because of the GPS Week Number Rollover which occured on April 6th, 2019, where the WN reached 1024 weeks and was reset to 0 weeks (max 10 bits field to manage the WN), then the XCsoar application is not managing this 2nd time GPS clocks needed to be reset.

Do you have an idea how to solve the issue?
-------------------------------------------

The XCsoar application shall correctly initiate de YYYY-MM-DD corresponding to WN0, which is now 2019-04-16.

lordfolken, 2022-12-21:
This is a firmware bug of your device. XCSoar refuses to workaround broken devices. 

Seems there is an update that fixes this that has been released in august 2022 for Samsung phones:
https://www.sammyfans.com/2022/08/22/samsung-releases-gps-fix-update-for-older-galaxy-phones/

closed 2022-12-24 by lordfolken

PascalLauper, 2022-12-31:
Thank you for the advise: I have upgraded my SAMSUNG S9+ with one UI 2,5 Android 10 from 1st March 2022 and then effectively the date is OK with XCsoar, for instance in Info/Status-Hours.

lordfolken, 2022-12-31:
Perfect. :) thx for the report.