XCSoar/XCSoar #1091: XCSoar 7.28 custom build for raspbery pi 0 src/ui/canvas/freetype/Font.cpp NextChar s.remove_prefix tries to remove to much from the string_view

mr-nice, 2022-12-30:
Hi all,
I got this problem on a custom build for my raspberry pi zero. Which is based on the meta-openvario xcsoar recipe.

At the end of the string view NextUTF8 returns a pair of 0,0 which is fine but afterwards it tries to s.remove_prefix(n.second - s.data()); which is wrong at least for my platform. 

This is the gdb output for the operation:

```c++
0x005a8e1c in ForEachGlyph<std::basic_string_view<char>&, Font::TextSize(tstring_view) const::<lambda(int, int, FT_GlyphSlot)> > (f=..., text=..., ascent_height=11, face=0xc61020)
    at /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp:251
251     in /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp
3: offset = <error: current stack frame does not contain a variable named `this'>
(gdb) n
260     in /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp
3: offset = <error: current stack frame does not contain a variable named `this'>
(gdb) 
/usr/include/c++/12.2.0/string_view:276: constexpr void std::basic_string_view<_CharT, _Traits>::remove_prefix(size_type) [with _CharT = char; _Traits = std::char_traits<char>; size_type = unsigned int]: Assertion 'this->_M_len >= __n' failed.

Thread 1 "xcsoar" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44      pthread_kill.c: No such file or directory.
3: offset = <error: current stack frame does not contain a variable named `this'>
```

I did a quit dirty fix for it and the patch is the following:

```c++
diff --git a/src/ui/canvas/freetype/Font.cpp b/src/ui/canvas/freetype/Font.cpp
index 29d7d61b3c..d80f8374b3 100644
--- a/src/ui/canvas/freetype/Font.cpp
+++ b/src/ui/canvas/freetype/Font.cpp
@@ -106,7 +106,10 @@ NextChar(tstring_view &s) noexcept
   return ch;
 #else
   auto n = NextUTF8(s.data());
-  s.remove_prefix(n.second - s.data());
+  if(n.second != 0)
+  {
+    s.remove_prefix(n.second - s.data());
+  }
   return n.first;
 #endif
 }
@@ -239,6 +242,8 @@ ForEachChar(tstring_view text, F &&f)
 
   while (!text.empty()) {
     const unsigned ch = NextChar(text);
+    if (ch == 0)
+           break;
     f(ch);
   }
 }
```

With that patch I'm able to run xcsoar -simulation without any problems.

So what are the expected values for the n.second - s.data() operation? And how would a proper fix look like?

Thanks in advance :)

MaxKellermann, 2022-12-31:
> At the end of the string view

No, at the end of the string_view, NextChar() will never be called:

>    while (!text.empty()) {
>      const unsigned ch = NextChar(text);

Don't you see? At the end, NextChar() is not reachable, because `text.empty()` will leave the loop.

I suppose you're dealing with malformed UTF-8, and this piece of code asserts that all UTF-8 is valid. It must have leaked from somewhere in here, and from there a runtime check is missing.

mr-nice, 2022-12-31:
Hey,
first of all thanks for the super quick response :) Yes you are right debugging late at night might not be the best idea ^^. I was so focused on the NextChar function that I completely overlooked the caller. But my problem remains.

The string is not mailformed but the exit condition of the while(!text.empty()){...} is evaluated differently at assembly level. I need to dig deeper into the assembly code. Which is always fun with optimized C++ code :D. So this could take some time. I need to find out if it is a logic, compiler, or a optimizer error.

This is the NextChar with examinated memory and displayed string. So the string looks good.

```c++
Thread 1 "xcsoar" hit Breakpoint 1, NextChar (s=...) at /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp:101
101     /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp: No such file or directory.
(gdb) info locals
n = <optimized out>
(gdb) print s
$1 = (tstring_view &) @0xbeffecc4: {_M_len = 18, _M_str = 0x6cbec0 <XCSoar_ProductToken> "XCSoar v7.28-Linux"}
(gdb) x/20xb s.data()
Cannot evaluate function -- may be inlined
(gdb) x/20xb 0x6cbec0
0x6cbec0 <XCSoar_ProductToken>: 0x58    0x43    0x53    0x6f    0x61    0x72    0x20    0x76
0x6cbec8 <XCSoar_ProductToken+8>:       0x37    0x2e    0x32    0x38    0x2d    0x4c    0x69    0x6e
0x6cbed0 <XCSoar_ProductToken+16>:      0x75    0x78    0x00    0x00
(gdb) x/20sb 0x6cbec0
0x6cbec0 <XCSoar_ProductToken>: "XCSoar v7.28-Linux"
0x6cbed3:       ""
0x6cbed4 <XCSoar_VersionStringOld>:     "Linux 7.28"
0x6cbedf:       ""
0x6cbee0 <XCSoar_VersionString>:        "7.28-Linux"
0x6cbeeb:       ""
0x6cbeec <XCSoar_VersionLong>:  "7.28"
0x6cbef1:       ""
0x6cbef2:       ""
0x6cbef3:       ""
0x6cbef4:       "Loaded profile from %s"
0x6cbf0b:       ""
0x6cbf0c:       "Failed to load profile"
0x6cbf23:       ""
0x6cbf24:       "Loading profiles"
0x6cbf35:       ""
0x6cbf36:       ""
0x6cbf37:       ""
0x6cbf38:       "Saving profiles"
0x6cbf48:       "default.prf"
(gdb) 
```

This is the end of the funciton where it should not reach NextChar with _M_len = 0 but it does.

```c++
ForEachChar<ForEachGlyph<std::basic_string_view<char>&, Font::TextSize(tstring_view) const::<lambda(int, int, FT_GlyphSlot)> >(FT_Face, unsigned int, std::basic_string_view<char>&, Font::TextSize(tstring_view) const::<lambda(int, int, FT_GlyphSlot)>&&)::<lambda(unsigned int)> > (f=..., text=...) at /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp:253
253     in /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp
2: text = {_M_len = 1, _M_str = 0x6cbed1 <XCSoar_ProductToken+17> "x"}
3: text = {_M_len = 18, _M_str = 0x6cbec0 <XCSoar_ProductToken> "XCSoar v7.28-Linux"}
(gdb) n

Thread 1 "xcsoar" hit Breakpoint 1, NextChar (s=...) at /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp:101
101     in /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp
1: s = (tstring_view &) @0xbeffecc4: {_M_len = 1, _M_str = 0x6cbed1 <XCSoar_ProductToken+17> "x"}
(gdb) 
109     in /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp
1: s = (tstring_view &) @0xbeffecc4: {_M_len = 1, _M_str = 0x6cbed1 <XCSoar_ProductToken+17> "x"}
(gdb) 
268           data() const noexcept
1: s = (tstring_view &) @0xbeffecc4: {_M_len = 1, _M_str = 0x6cbed1 <XCSoar_ProductToken+17> "x"}
(gdb) 
110     /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp: No such file or directory.
1: s = (tstring_view &) @0xbeffecc4: {_M_len = 1, _M_str = 0x6cbed1 <XCSoar_ProductToken+17> "x"}
(gdb) 
117     in /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp
1: s = (tstring_view &) @0xbeffecc4: {_M_len = 1, _M_str = 0x6cbed1 <XCSoar_ProductToken+17> "x"}
(gdb) 
119     in /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp
[... 413 lines cut ...]
231           empty() const noexcept
2: text = {_M_len = 0, _M_str = 0x6cbed2 <XCSoar_ProductToken+18> ""}
3: text = {_M_len = 18, _M_str = 0x6cbec0 <XCSoar_ProductToken> "XCSoar v7.28-Linux"}
5: x/i $pc
=> 0x5a9074 <_ZNK4Font8TextSizeESt17basic_string_viewIcSt11char_traitsIcEE+240>:        bne     0x5a8fe8 <_ZNK4Font8TextSizeESt17basic_string_viewIcSt11char_traitsIcEE+100>
(gdb) 
0x005a8fe8      250     /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp: No such file or directory.
2: text = {_M_len = 0, _M_str = 0x6cbed2 <XCSoar_ProductToken+18> ""}
3: text = {_M_len = 18, _M_str = 0x6cbec0 <XCSoar_ProductToken> "XCSoar v7.28-Linux"}
5: x/i $pc
=> 0x5a8fe8 <_ZNK4Font8TextSizeESt17basic_string_viewIcSt11char_traitsIcEE+100>:        mov     r0, r7
(gdb) 
0x005a8fec      250     in /usr/src/debug/xcsoar/7.28-r0/src/ui/canvas/freetype/Font.cpp
2: text = {_M_len = 0, _M_str = 0x6cbed2 <XCSoar_ProductToken+18> ""}
3: text = {_M_len = 18, _M_str = 0x6cbec0 <XCSoar_ProductToken> "XCSoar v7.28-Linux"}
5: x/i $pc
=> 0x5a8fec <_ZNK4Font8TextSizeESt17basic_string_viewIcSt11char_traitsIcEE+104>:        bl      0x5a8a24 <NextChar(tstring_view&)>
(gdb) x/10xb 0x5a8fe8 
0x5a8fe8 <_ZNK4Font8TextSizeESt17basic_string_viewIcSt11char_traitsIcEE+100>:   0x07    0x00    0xa0    0xe1    0x8c    0xfe    0xff    0xeb
0x5a8ff0 <_ZNK4Font8TextSizeESt17basic_string_viewIcSt11char_traitsIcEE+108>:   0x00    0x10
(gdb) x/10cb 0x5a8fe8 
0x5a8fe8 <_ZNK4Font8TextSizeESt17basic_string_viewIcSt11char_traitsIcEE+100>:   7 '\a'  0 '\000'        -96 '\240'      -31 '\341'      -116 '\214'     -2 '\376'       -1 '\377'       -21 '\353'
0x5a8ff0 <_ZNK4Font8TextSizeESt17basic_string_viewIcSt11char_traitsIcEE+108>:   0 '\000'        16 '\020'
(gdb) 
```

MaxKellermann, 2023-05-03:
You did not post any data about the string being processed; in the last post, you set a breakpoint, but all that data is completely irrelevant, because there was no crash here. 4 months later and no more information, closing this.

closed 2023-05-03 by MaxKellermann