XCSoar/XCSoar #1092: driver for Becker AR62xx

Blaubart, 2022-12-31:
Here is a driver published for the radio Becker AR62xx. What do think to implement it to the official repository?
https://github.com/eku60/XCSoar/tree/ar62xx_radio/src/Device/Driver

We testet the driver and it works fine.

MaxKellermann, 2022-12-31:
Where's the protocol documentation?
There's a policy that new code to support commercial products will only be merged into XCSoar if the vendor publishes the protocol documentation non-discriminatory and for free. I think that conflicts with Becker's copyright policy, which is hostile towards their customers - or am I mixing up Becker with somebody else?

Blaubart, 2022-12-31:
you are right, Becker dosen't allow to publish protocol documentation :-(

MaxKellermann, 2022-12-31:
Okay then, no support for Becker in XCSoar. Sorry.
Don't buy Becker.

Blaubart, 2022-12-31:
OK ;-)

closed 2023-01-06 by lordfolken

iltis42, 2025-03-31:
> Okay then, no support for Becker in XCSoar. Sorry. Don't buy Becker.

Yeah, that's also what i recommend everybody, hopefully they change their policy at times

lordfolken, 2025-03-31:
Apossiblity is to install an air avionics  display, which has very good pulbic 3rd party documentation. They support becker afaik. And there is a good driver for the Air Avionics Display in XCSoar.