XCSoar/XCSoar #1099: Pilot Event Marker

RichardFrawley, 2023-01-23:
I have just been setting up PEV on XCSoar for upcoming comps…

I have set both PEV Start Wait Time, and PEC start window at 10 Mins

When I go to NAV>PILOT EVENT ANNOUNCE the Start Open comes up with a “PILOT EVENT ANNOUNCED” dialogue box and the Start Open NAV Box comes up with a timer counting down, but starting with the time greater than 10 Mins.

Start Reach box says 9:36 and Start Open box says 10:36  

why is it showing these numbers ?

I assume start reach is that I would arrive at the start line with 9:36 min to go???? 

By why is start open box showing 10:36???

The manual did not seen to have anything useful to say on this.

kobedegeest, 2023-01-23:
Think it might be rounding "error". Start line open time seems to be accurate to the minutes hence the extra seconds in the count down.

kobedegeest, 2023-01-23:
Yes i was just comenting to give possible cause of the issue. I noticed a more annoying issue however with the pev when a second pev is done after the startline was already open the wait time is for some reason = time startline has been open + wait time.

RichardFrawley, 2023-01-23:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Yes, its seems to be overly complex

Just 2 simple count down counters would have been perfect orc even the same colour change one one.

And start the whole 2 stage count down with a touch to the Nav box which also creates the PEV .

The simpler from the pilot perspective the better

Richard Frawley  \  CXO  \  TZ Limited
+61 425 205 274

kobedegeest, 2023-01-23:
Never mind my previous comment think it was just because i was testing in sim mode and had it running in background and then time gets messed up it seems. 

Thus only problem is that start time is only defined up to minutes.

RichardFrawley, 2023-01-23:
Hi

There is also a time difference between the info boxes. There is no description in the manual about how start reach is supposed to work. I assume when you press the PEV, it’s is showing what time you would have left when you reach the start, but, as than can be a line often up to 10min, it’s actually not of much help and just confuses things. It would be better (I have asked at least 10 pilots) if both were just simple timers as per my previous comments. Many thanks

kobedegeest, 2023-02-22:
I had a look around in the code and from what i can tell the StartConstraints.open_time_span is a RoughTimeSpan which is only accurate up to minutes. With the start and end being of type RoughTime. 

In XCSoar/src/Input/InputEventsActions.cpp the wait time is increased by one minute in order to make sure start is open when XCSoar says it is but it seems like this implementation can have you start when it is already closed. Reducing the start window by one would ensure vallid start but decrease the start window. 

Since competitions where this is used use it in seconds accuracy this seems like the wrong data type for this parameter. So think this is more a bug than lack of documentation.

Changing the data type would solve it i think however as far as i can see there appears to be no parameter type like RoughTime which is accurate up to seconds implemented at the moment. But maybe i am just not seeing it? 

Is there a reason that stored times are only minute accurate? performance wise or memory wise? 

The way i see it there two options  change the start window as well to ensure valid start and add documentation to inform pilots of this inaccuracy. Or better make new data type AccurateTime and AccurateTimeSpan and use this for the StartConstraints.open_time_span.

(note i am no expert with c++ or github but like to learn and start to help out where possible.)

RichardFrawley, 2023-02-22:
I have been now at 3 coaching and competition events this season and spoken to many XC Soar users and they all agree that the current PEV is unusable.

They just want simple and accurate (to the second) countdown timers

lordfolken, 2023-02-26:
@kedder could you have a look please? I think this would be great if it was fixed this season.

kedder, 2023-02-26:
The PEV start is pulled to the next round minute. This is an unfortunate limitation of "start open" time setting in XCSoar - it is set to the round minute. However, since XCSoar opens the time slightly later than actual period, you will not start before the period has passed.

> Start Reach box says 9:36 and Start Open box says 10:36

Start reach shows when you need to turn towards the start to reach it in time, "Start Open" shows when the start actually opens. They are not supposed to show the same time.

kobedegeest, 2023-06-05:
> The PEV start is pulled to the next round minute. This is an unfortunate limitation of "start open" time setting in XCSoar - it is set to the round minute. However, since XCSoar opens the time slightly later than actual period, you will not start before the period has passed.

You will not start before the window opens but I believe with the current implementation you can unknowingly start after it has closed. Which is just as bad. Since the closing time does not get "rounded" down.

But either way this implematation is bad and pilots are not informed over this shortcomming. Even if the closing time is within "safe" margin, if the open time is like 3 minutes and in worst case you lose 1 of that that is a third of your start window you lose.

barreralu, 2024-09-18:
@kedder  could you have a look please? I think this would be great if it was fixed this season.
Regards From Argetina!

andrestotorica, 2025-04-18:
I sent a PR that fixes the rounding to minutes issue and gives second accuracy. However, after testing and thinking about it, I feel this feature is still not ready for contest usage.

In the current implementation, shooting a PEV sets the task start & end times, which effectively disallows XCSoar from computing a valid start outside that window. However, in a contest task the pilot may cross the start at any time, even without a PEV, only at the cost of a penalty at the end of the day. I see very plausible that someone may decide to start e.g. a few seconds outside the window. Or, if the competition officials delay the start, the PEV becomes not mandatory anymore. 

In such a case, the pilot would be unable to just compute a start without fiddling with the Task Manager to clear constraints before starting. Or, if the start was already already crossed, there is no way of retroactively making the start valid (I hink, will investigate a bit further).

In any case, I think the Start Open and Start Close times should be a separate thing from the PEV start window, and enforced separately. I will let the current PR go it's course as it still has value, but explore how the feature overall can be improved. Any ideas or suggestions are welcome.

andrestotorica, 2025-04-25:
Thanks @kobedegeest for the input

> In settings Task defaults > Task rules there are settings for a margin on the start speed and alt but these seem to also not work or do anything. 

These seem to work as expected to me, I never used them before but the code is there and I ran a quick simulation to check  the margins are considered. Keep in mind the margins are not shown when setting up the task, but they are used internally when checking if the start line crossing was valid. Let me know if you know of any issue with them not working.

> could do a similar thing for the Pev window (maybe not a margin but just an allow start outside start open window). 

My thinking here was that, in a competition that requires the PEV procedure (such as the WGC) there is no other start time requirement. I'm not aware of tasks that use the fixed time start window, but I guess they'll probably won't use a PEV. 
So we could have XCSoar handle them separately: for times set in the task rules, a hard start window that is enforced (no change). For the PEV press, a soft start window which is only informative via the infoboxes. That way we avoid a new setting that could be potentially misconfigured.

andrestotorica, 2025-05-15:
> Yeah also just did quick check with the height, tested again just now. Does not seem to work for me. Crossing startline above max alt but within the margin does not start the task ...

Hmm that's weird, I don't think the settings can be out of sync. Have you considered other constraints preventing the start? AGL vs MSL setting for max alt? You can also try to produce an igc, task file and a profile where it does not work to reproduce the issue and share.

> Is there typically not still the start open time (like x minutes after last takeoff)? But no need to overcomplicate things.

Yea of course, but that is also subject to change from the organization, announced on the radio, etc, and would be outside of XC Soar (or any other navigator/glide computer) scope.