XCSoar/XCSoar #1100: Not compiling for Mac OS X

PeterNSteinmetz, 2023-01-23:
While the developer pages talk about building for Mac OS X using the homebrew system (which sounds like an excellent idea), if I start the build as instructed with 'make dmg' the build fails with what appears to the be loader, ld, not able to process a -lSystem flag when building libsodium. This seems unnecessary since that library is available in homebrew. Full log attached. Also XCSoar/output/OSX64/lib/build/libsodium-1.0.18/config.log attached.

I tried just removing the libsodium line in the thirdparty.py code, but this build system is rather complex. 

Is it the intention to continue to be able to build on MacOS X? Anyone with more experience with this build system want to try and get it building again on that platform? 

[config.log](https://github.com/XCSoar/XCSoar/files/10476652/config.log)
[bugListing.txt](https://github.com/XCSoar/XCSoar/files/10476655/bugListing.txt)

lordfolken, 2023-01-28:
have a look at the native github worklfow: 
https://github.com/XCSoar/XCSoar/blob/2e3e49be75cae354fc2ec2fbc524108a5fae3559/.github/workflows/build-native.yml#L420-L470

This is what is currently in use in the pipeline.

PeterNSteinmetz, 2023-01-31:
Thanks. I just had a look at that. So why is thirdparty.py running in my build and then failing? 

That seems to be what happens when I run the command 'make dmg' as suggested in the docs.

lordfolken, 2023-02-05:
OSx is just a compile target for us, so that we see that we don't add to platform specific code. None of the developers actually has apple hardware, nor is it currently a goal to support that ecosystem. The person that built the port  unfortunately no longer maintains it.

PeterNSteinmetz, 2023-02-05:
Might then be a good idea to remove references to MacOS X from the documentation and web site until a new maintainer is found.

closed 2023-02-05 by PeterNSteinmetz