XCSoar/XCSoar #1105: Connection last about 5 seconds

Dave-Baxter, 2023-01-30:
EXCEPTION: Send error An existing connection was forcible closed by the remote host

Log Name:      Application
Source:        Application Hang
Date:          1/29/2023 2:45:10 PM
Event ID:      1002
Task Category: Hanging Events
Level:         Error
Keywords:      
User:          SYSTEM
Computer:      Unit-5
Description:
The program XCSoar v7.28.exe version 0.0.0.0 stopped interacting with Windows and was closed. To see if more information about the problem is available, check the problem history in the Security and Maintenance control panel.
[image]
[image]
[image]
[image]
[image]

lordfolken, 2023-01-30:
Can you please fill out the bug template?

A nmea dump from this program would be interesting. Can you turn on nnea logging and attach the dump? Although i suspect its more a connection handling issue. Xcsoar is on the recieving end here though, so im not sure how much can be done.
I would also ask the vendor to investigate as they seem to be rargeting xcsoar specifically.

lordfolken, 2023-01-30:
I also see that you are not using the correct driver. The program is sending xcvario protocol, but the generic driver is in use.

Dave-Baxter, 2023-01-30:
I'll try that this evening. My system is brand new dual boot with identical software on both. Win10Pro everything works fine and I've been using it for a couple weeks. With Win11Pro it throws this error/issue and the connection breaks and the map just sits at the location it adjusted to during the 5-10 second period it was connected.

Dave-Baxter, 2023-01-30:
XCSoar hangs after attempting to close it, every single time you try to close it, it hangs and has to be forced to close. This only happens on Win11 Pro on the Win10 Pro side of my dual boot it closes normally and run perfectly.

Dave-Baxter, 2023-01-30:
Thanks for taking time out to respond I will keep an eye on this tread I created.

Dave-Baxter, 2023-01-31:
Greetings Phillipp,

Running Windows 10 it works great. Windows 11 the program stops responding
when you try to close it. Requiring the task manager.

Log File:
2023-01-30 23:36:59Z: log copied to clipboard
2023-01-30 23:34:42Z: Kinetic Assistant connected
2023-01-30 23:27:52Z: log copied to clipboard
2023-01-30 23:26:22Z: NMEA Server: 1
2023-01-30 23:26:15Z: NMEA Server: OK
2023-01-30 23:25:34Z: NMEA Server: 1
2023-01-30 23:25:18Z: NMEA Server: OK
2023-01-30 23:24:29Z: NMEA Server: 1
2023-01-30 23:23:47Z: ALBATROSS membership is active
2023-01-30 23:23:46Z: NMEA Server: OK
2023-01-30 23:23:46Z: Kinetic Assistant+ 0.17.3.0 loaded
2023-01-30 23:23:46Z: MSFS Path: C:\Users\david\AppData\Roaming\Microsoft
Flight Simulator\
2023-01-30 23:23:46Z: NMEA Server: OK
2023-01-30 23:23:46Z: NMEA Server: START
2023-01-30 23:23:46Z: Community Path:
C:\Users\david\AppData\Roaming\Microsoft Flight Simulator\Packages\Community

Here's a video of the how during closing it "Stops Responding"
https://photos.app.goo.gl/MwDakXbo4WDNnMRQ9

I look forward to hearing more about this. Reminder, my machine is brand
new with a dual boot Win10 Pro & Win11 Pro on two seperate M.2 SSD's with
exact same software on both.
The Win10 runs these two apps and the process flawlessly. Go figure!

Dave Baxter
461 East Mountain Rd.
Newport, NH 03773
Cell: (603) 477-6185
Google Voice: (802) 448-0260
***@***.***

garcialcg, 2023-02-01:
I downloaded XCSOAR V 7.28 last November in my Samsung tablet A7 and in a cell phone Samsung galaxy S 20. They were working well in both until 2 days before the app is disconnected alone after few seconds on just in the tablet. Some time an error message requesting remove bugs is prompted. However, the same version is still working well in the cell phone.
What could I do to fix this?
Thanks

Dave-Baxter, 2023-02-03:
My beloved XCSoar "Fails to respond" when closing and requires CLT-ALT-DEL
to close. It was working great. It's also now failing to get the GPS
information through kinetic assistant from Microsoft Flight Simulator. I
have a dual boot system and this was always happening on Windows 11 but
Windows 10 was working fine now Windows 10 is doing the same thing. Any
help would be great as to where to go look, try.

Regards Dave

Dave-Baxter, 2023-02-03:
Hello again Phil,

XCSoar on my Windows 10 install (it was working fine all week until last
night)  is now doing what windows 11 was doing which is "Failing to
respond" while try to connect to Kinetic Assistant and grab the GPS data,
and then NOT RESPONDING afterwards while trying to close the application.
The message log in Kinetic Assistant reads the following: Connect has been
terminated.

Hoping for a resolution as I love using it.

Regards,
Dave

lordfolken, 2023-02-04:
I have multi hour simulation flights with windows 10 (condor simulator). I even installed a virgin windows 11 in a vm and cannot reproduce any of  your described problems. 

Does it work when you don't have this simulation addon running? 
Did you allow XCSoar in the application firewall?
It works fine in  SIM mode? 
Potentially the software's stream isn't good. 
For that i really need a nmea dump.  Configuration -> Setup -> Logger -> NMEA Logger

That will produce a nmea file in the logs folder which you can attach to this ticket.

Dave-Baxter, 2023-02-05:
2023-02-05 03:59:29Z: log copied to clipboard

2023-02-05 03:58:09Z: EXCEPTION: Can't process callback Cannot access a
disposed object.

Object name: 'System.Net.Sockets.Socket'.

2023-02-05 03:58:09Z: NMEA Server: OK

2023-02-05 03:58:09Z: NMEA Server: OK

2023-02-05 03:58:09Z: NMEA Server: START

2023-02-05 03:58:09Z: NMEA Server: STOP

2023-02-05 03:54:57Z: ALBATROSS membership is active

2023-02-05 03:54:56Z: NMEA Server: OK

2023-02-05 03:54:56Z: Kinetic Assistant+ 0.17.3.0 loaded

2023-02-05 03:54:56Z: MSFS Path: C:\Users\david\AppData\Roaming\Microsoft
Flight Simulator\

2023-02-05 03:54:56Z: NMEA Server: OK

2023-02-05 03:54:56Z: NMEA Server: START

2023-02-05 03:54:56Z: Community Path:
C:\Users\david\AppData\Roaming\Microsoft Flight Simulator\Packages\Community

Dave-Baxter, 2023-02-05:
Now this:

2023-02-05 04:04:24Z: log copied to clipboard

2023-02-05 04:04:06Z: toggleConnect1 Could not connect to Flight Simulator:
Error HRESULT E_FAIL has been returned from a call to a COM component.

2023-02-05 03:59:29Z: log copied to clipboard

2023-02-05 03:58:09Z: EXCEPTION: Can't process callback Cannot access a
disposed object.

Object name: 'System.Net.Sockets.Socket'.

2023-02-05 03:58:09Z: NMEA Server: OK

2023-02-05 03:58:09Z: NMEA Server: OK

2023-02-05 03:58:09Z: NMEA Server: START

2023-02-05 03:58:09Z: NMEA Server: STOP

2023-02-05 03:54:57Z: ALBATROSS membership is active

2023-02-05 03:54:56Z: NMEA Server: OK

2023-02-05 03:54:56Z: Kinetic Assistant+ 0.17.3.0 loaded

2023-02-05 03:54:56Z: MSFS Path: C:\Users\david\AppData\Roaming\Microsoft
Flight Simulator\

2023-02-05 03:54:56Z: NMEA Server: OK

2023-02-05 03:54:56Z: NMEA Server: START

2023-02-05 03:54:56Z: Community Path:
C:\Users\david\AppData\Roaming\Microsoft Flight Simulator\Packages\Community

lordfolken, 2023-02-05:
Please see with the manufacturer of this Kinetic Assistant.

closed 2023-02-05 by lordfolken