XCSoar/XCSoar #1106: XCREMOTE integration

XCNav, 2023-02-04:
XCSoar version
--------------

all

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Native integration of XCRemote

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
As we have sold more than 60 XCRemote BT remote controll sticks with a growing demand, it would be a good Idea to implement it in the XCSoar source code. Right now we are working with a .xci file. XCREMOTE mainly picks up native XCSoar basic commands but also picks up some special NMEA commands of the XCVario emulating the Rotary Knob of XCVario as to make it fully controllable via XCRemote without actually touching the Vario.

I would like to have the XCRemote selectable in the settings of XCSoar to circumnavigate the .xci procedure.

Here you can see present .xci file

https://github.com/XCNav/XCRemote/blob/main/xcremote-XCV_V2.2.xci

lordfolken, 2023-02-06:
https://github.com/lordfolken/XCSoar/tree/xcitype 
will propose ist as soon as i have the xci in the https://github.com/XCSoar/xcsoar-data-content

lordfolken, 2023-02-08:
depends on https://github.com/XCSoar/xcsoar-data-content/pull/308

iltis42, 2023-02-08:
The xci interface provides a high layer of abstraction with extended capabilities e.g. the trigger by SendNMEAPort[1|2] is quite cool, hence lacks a bit documentation, afaik any documentation for the last think mentioned ;), so the wish to get around the .xpi story for the enduser at all i can understand. 
The file in the data content is see as the first step, and there is more support is thinkable IMHO. For me in XCSoar i see this also either as a downloadable thing than an integral part of XCSoar. Today, when you select this Events file in the Language, Input Tab, the dialog is empty and nothing is provided to fill it up unless you copy manually the file in the XCSoarData folder (wherever this is, we have learned on android this might even be invisible without workarounds for your file managers). 
In my opinion a "Download" button in the Events dialog could close this gap to the xcsoar data content where those files are:
https://github.com/XCSoar/xcsoar-data-content/tree/master/data/content/xci/global
and once the device is online all .xpi's from data content may be downloaded when the "Download" is pressed in there, and if we call the file XCREMOTE.xpi, the dialog may list just XCREMOTE (beside other to come of course), i would say that's the ticket.

closed 2023-02-10 by MaxKellermann