XCSoar/XCSoar #1112: Add NMEA GPS information to ACD driver

XCNav, 2023-02-11:
XCSoar version
--------------

All versions

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Presently the ACD driver only supports Altitude and Com data exchange between XCSoar and ACD

Add GPS Position information

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
The ADC supports an autotune function to dial in the frequency of the nearest alternate airport as standby freq. Therefore it needs a gps position. As devices running  XCSoar naturally provide GPS information it would be usefull to pass this as well to the ACD so no second data input from a gps source is needed.
 See above

hjr, 2023-02-11:
I like this idea. Need to be configurable, because people might use other gps sources connected to ACD.

DanD222, 2023-02-11:
It would also be great to extend the functionality of say the VT-01 transponder to ADS-B using GPS positional data, so an option to send GPS position information could probably be useful for the Generic driver as well? (or is the better way to do this using the xci system?)

Scumi, 2023-02-13:
> I like this idea. Need to be configurable, because people might use other gps sources connected to ACD.

You can disable processing of GPS data on each port in the ACD config, so there would be no need to do that on XCSoar side.

I restarted work on sending GPS data to the ACD from XCSoar based on my old PR #170. Not sure if I'm more successful this time, but refactoring goes quite well for now, so lets see...

XCNav, 2023-03-23:
Is there still a device needed. I can organize one by AirAvionics.
Just contact me via info@xcnav.de

Scumi, 2023-03-23:
For what? I have opened a PR, which I tested on my device. PR is linked right above your comment. It just needs review and merge.

Scumi, 2023-04-25:
@XCNav you can close this now, PR is merged.

lordfolken, 2023-04-25:
Closing as implemented.

closed 2023-04-25 by lordfolken

XCNav, 2023-04-25:
Thanks to all!