XCSoar/XCSoar #1116: High DPI icons

lordfolken, 2023-02-14:
XCSoar version
--------------
v7.28

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Current mobile screens have resolutions of up to 650dpi. 
XCSoar's icons currently scale to maximum of 160 DPI.

Ideally we can use the SVGs to generate higher DPI icons.

August2111, 2023-02-19:
Before we do this bmp/png to svg conversion in the program code, it is in my opinion faster to adjust the critical (too small) bitmaps. In my opinion, this includes, for example, the 'mode' icons cruise, climb and finalglide. Or do you think that all icons are generally displayed too small?

lordfolken, 2023-02-19:
It concerns a lot of icons displayed on the map, the flight mode icons you mentioned, but also the icons in the task manager, flarm status, gps, terrain collsion icons, metar icons etc. 

Unfortunately its a little more complex than just upscaling the icon:
 In general all bitmaps are generated during the build process from svgs here (in several different dpi versions)
https://github.com/XCSoar/XCSoar/blob/8f1cbf368cf591e03a83bf5d90cb4e26421bb1c3/build/resource.mk#L36-L68

The source svgs are here https://github.com/XCSoar/XCSoar/tree/master/Data/icons 

Additionally they are assigned a resource id in src/Resources.hpp (currently in 2 resolutions standard and "hd"). 

Then each resource is loaded at the appropriate place for example:
https://github.com/XCSoar/XCSoar/blob/8f1cbf368cf591e03a83bf5d90cb4e26421bb1c3/src/Look/MapLook.cpp#L84-L86

The load resource function seems to be defined here, and seems to have some ability to stretch icons. Maybe its artificially limited, or another issue:
https://github.com/XCSoar/XCSoar/blob/8f1cbf368cf591e03a83bf5d90cb4e26421bb1c3/src/ui/canvas/Icon.cpp#L61

August2111, 2023-02-19:
I know the complexity of creating the icon bitmaps pretty well: I switched the make process to CMake for myself, because it's easy for me to handle - and I had to deal with the bitmap resources quite intensively. (But after Max made it clear to me about 1.5 years ago that he doesn't prefer CMake as a build process, I just saw it as my own hobby...) For me, the question also arises as to how the various graphics systems deal with the resources again, Linux has a completely different UI setup than Android - and than Windows, you always have to be extremely careful at this point that you don't destroy the other targets with a meaningful change in one target ;-( Me would like to have a uniform system (OPEN_GL vs. MEMORY_CANVAS vs. GDI) at this point before getting even more deeply involved in the various implementations.

lordfolken, 2023-02-20:
Well refactoring is always work, and i don't think changing the build system would reduce much of the complexity. 
My first question is whether we still need BMP? 
As far as I know all platforms should now support PNG. That would reduce the amount of paths quite considerably already. 
Then we could create high res icons and then downscale with a function, instead of trying to have many different versions of the same pic.

August2111, 2023-02-20:
> Well refactoring is always work, 
That's clear
> i don't think changing the build system would reduce much of the complexity.
I just wanted to explain why I've already dealt with the whole resource problem quite intensively... By the way, I need the CMake for my preferred tool VisualStudio - I'm having a hard time with the .mk files ;-(
> My first question is whether we still need BMP?
The BMPs are still needed under Windows (target PC or WIN64), they would first have to be eliminated there ;-(
Otherwise, it's definitely a good idea to reduce the many specially created images, I think the small formats are probably no longer used...

lordfolken, 2023-02-20:
There is a comment here:
https://github.com/XCSoar/XCSoar/blob/8f1cbf368cf591e03a83bf5d90cb4e26421bb1c3/src/ui/canvas/gdi/GdiPlusBitmap.cpp#L54-L56 (i think its by you.) that GdiPlus would support png?

lordfolken, 2023-02-20:
https://en.wikipedia.org/wiki/Graphics_Device_Interface#Windows_XP

"With the introduction of [Windows XP](https://en.wikipedia.org/wiki/Windows_XP), GDI+ complemented GDI. GDI+ has been written in C++. It adds anti-aliased 2D graphics, floating point coordinates, gradient shading, more complex path management, intrinsic support for modern graphics-file formats like [JPEG](https://en.wikipedia.org/wiki/JPEG) and [PNG](https://en.wikipedia.org/wiki/Portable_Network_Graphics), and support for composition of [affine transformations](https://en.wikipedia.org/wiki/Affine_transformation) in the 2D view pipeline. GDI+ uses [RGBA](https://en.wikipedia.org/wiki/RGBA) values to represent color."

I think it would be safe to deprecate platforms before windows XP in 2023?

August2111, 2023-02-23:
> (i think its by you.) that GdiPlus would support png?

That's right, I introduced that to be able to load weather images from PCMet. However, there is currently no function that can load *.png resource images in GDI+, so all the *.bmp bitmaps are still necessary (only for GDI - and therefore only for Windows).
With a GDI+ ResourceLoad function for *.png you can do without all *.bmp... I would do that as a first step for this adaptation!

MikeBravo15, 2023-03-18:
I have made a test whith just upscaling the icons. I used higher resolution icons increased to 32x32 in resources.mk together  with my modified icons, where all icons are 32x32px, however map icons only 2/3 of the whole area. Results on my Pixel4a nad Nokia8 look more than acceptable - menu and mode icons are big and map icons aru just right. Can you please test the build at https://github.com/Sundown3867/XCSoar/actions/runs/4457203274 and report any issues on other devices with different display resolutions? Thank you for any feedback

lordfolken, 2023-03-19:
So the icons should only be changed, relative to each other.  I.e. this icon is 1 so that icon should be factor 1.5. 
The pixel value as such is not useful, as these are SVGs. They will scale to any size. 

Unfortunatly XCSoar can't use SVGs directly, and needs appropriate bitmaps. So all the svgs are converted to PNGs by
rsvg-convert in build/resource.mk

XCSoar currently only knows 100dpi and 160dpi. Modern smart phones do 650Dpi.

lordfolken, 2023-03-19:
rsvg-convert can do scaling to dpi:

```man
   OPTIONS
       -d --dpi-x number
              Set the X resolution of the image in pixels per inch.  RSVG's current default is 90 DPI.

       -p --dpi-y number
              Set the Y resolution of the image in pixels per inch.  RSVG's current default is 90 DPI.
```

So once the icons are all in correct size and viewport (#1155), we can replace the fixed pixel with, with a dpi value. here:
https://github.com/XCSoar/XCSoar/blob/30e5119e5a56ba927deb9b82575f0096e12fd670/build/resource.mk#L42-L46

lordfolken, 2023-03-19:
There is some information on icons in the dev manual: https://xcsoar.readthedocs.io/en/latest/mapfile.html#adding-new-icons

lordfolken, 2023-03-19:
https://developer.android.com/training/multiscreen/screendensities#TaskProvideAltBmp
Some icon sizes as they should be depending on the dpi of the screen.

August2111, 2023-03-19:
@lordfolken  is absolutely right, the pixel size in the SVG file can be scaled up and down as required (DPI)...
It is more problematic that every (target) system deals with it differently: Windows still needs bitmaps (.bmp) with the GDI, Linux with MemoryCanvas works with Png's, Android with OpenGL can probably deal directly with .svg...
I'm pretty far along in eliminating the *.bmp files now, because GDI+ can handle PNG too! This simplifies the bitmap generation enormously!
I would also prefer if all systems only dealt with SVG, but I (or someone else) will probably have to do a few more loops :)

lordfolken, 2023-03-19:
> I'm pretty far along in eliminating the *.bmp files now, because GDI+ can handle PNG too! This simplifies the bitmap generation enormously!

That would be Awesome[tm].

MaxKellermann, 2024-02-28:
The discussion about PNG is completely off-topic here (has nothing to do with providing higher-res icons), and is also misguided. Yes, GDI+ can load PNG **files**, but this is not about loading **files**, it's about loading Win32 resources, and that part of the Win32 API only supports BMPs. Sure, you could write code that stores PNG files in binary resources and then parse that resource, but ..... why bother? Windows is the only target that uses BMPs, all the others use PNG.

About using SVG: sounds like a good idea, but unfortunately rendering SVG is extremely complex. Loading librsvg, for example, loads 20 MB of code into the program. Huh, what? Yes, that's true. 20 MB of code. SVG is that complex, and implementations are that much bloated. Unfortunately, SVG is the only vector graphics standard, so we have no choice.

We could, however, easily render more variants to PNG.

closed 2024-03-01 by MaxKellermann