XCSoar/XCSoar #1118: OpenAIR Parser: Parse the AY field

lordfolken, 2023-02-15:
What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Parse the AY field and display it in the Airspace popup when selecting an airspace

```
AY = Airspace Type (extends AC which should now only be used for class)
```

Example Airspace:
```
AC E
AY TMA
AN TMA BALE T1 HX
AI 62e10d61d0d055fccf50f375
AL 5200ft MSL
AH 5700ft MSL
DP 47:25:52 N 007:46:41 E
DP 47:19:12 N 007:51:31 E
DP 47:17:28 N 007:48:42 E
DP 47:19:05 N 007:26:21 E
DP 47:24:44 N 007:22:28 E
DP 47:22:60 N 007:39:30 E
DP 47:25:52 N 007:46:41 E
```

ubx, 2023-02-16:
Do you mean something like that?
[image]

lordfolken, 2023-02-17:
Daah gseht guet us! (This looks good!)

closed 2023-03-21 by MaxKellermann