XCSoar/XCSoar #1122: Changing number of markers changes Home field

kobedegeest, 2023-02-19:
XCSoar versions having and not having the problem
-------------------------------------------------
tested in v7.28

Expected behavior
-----------------

One can use the marker function to add and remove markers without it to cause the home field to change. 

Actual behavior
---------------

When a marker is added or existing markers are removed, basicly when number of markers changes. Next time you start up XCSoar the home field is different.

Steps to reproduce the behavior
-------------------------------

Start up XCSoar either in fly or sim mode --> add a marker --> close XCSoar --> open XCSoar fly or sim mode and check home field it will be different airfield. (Checking in sim mode is easiest as plane starts at home field).

Do you have any idea what may have caused this?
-----------------------------------------------

From my testing it looks like home field is stored as some kind of list index and markers are added at the top of the list changing all the indices without changing what index is used for home thus changing home field.

Do you have an idea how to solve the issue?
-------------------------------------------

Add markers at bottom of list or different list or store home field by name or coordinate.

ubx, 2023-03-13:
Traced/debugged down the problem a bit: after adding a marker, the content of the home waypoint, with the same id (here 1508) has changed.
**Marker Drop**
[image]

Restart XCSoar and **Marker Drop** again:
[image]

kobedegeest, 2023-03-14:
Yeah my guess is that the markers are added at the top of the list that stores the waypoints moving everything down.
Maybe the coordinates of the home field can be stored instead of id (or name as in your PR)? 

But imho storing name should be fine home field will most of the time have the name of the airfield.

lordfolken, 2023-03-15:
There are flags for the waypoints. We should assign the flag home to the waypoint and use it at runtime. Safe it at shutdown.

kobedegeest, 2023-03-15:
Maybe to clarify a bit it does not change during runtime. Only changes after restart where the id is used but the correct waypoint now has a different id. 
I noticed that in the profile file the HomeLocation is already stored as well as the HomeWaypoint (which is the id). 
So maybe something similar to how tasks are stored and loaded can be done.

kobedegeest, 2023-08-23:
I had a new look at the code to see where the waypoints get read. Solving the problem is as easy as swapping some lines of code. 
So in XCSoar/src/Waypoint/WaypointGlue.cpp, 4 waypoint files are read in and the user.cup where the markers are stored is read first where it would be beter to read it last. so if lines 79-81 are moved down to line 100 this issue would be solved and id would still be used to store the home so no problems with uniqueness or anything. 

And since i think it makes sense that home does not stay consistent when you change one of the other files i think this solution should be good.

PR #1287 (2023-09-22): Change order in which waypoint files are read
this closes issue #1122 because now markers will be last in the waypointlist meaning the id of the other waypoints does not change.

Brief summary of the changes
----------------------------
Moved loading the user.cup file (which contains the markers) down to after all the other waypoint files have been read. 

Related issues and discussions
------------------------------
Closes #1122
closed 2023-09-22 by MaxKellermann