XCSoar/XCSoar #1123: enableNotificationQueue.add(c); never reached in BluetoothSensor.java

ygaillard, 2023-02-19:
XCSoar versions having and not having the problem
-------------------------------------------------

File android/src/BluetoothSensor.java in version 69f7740b312ad6c470d737b0d6ab14628cdd8191
Function `  private void enableNotification(BluetoothGattCharacteristic c)` never reaches ` enableNotificationQueue.add(c);`

Expected behavior
-----------------

If `doEnableNotification(c)` return true. The characteristic should be added to the notification queue.

Actual behavior
---------------

On start the `currentEnableNotification`is null. Then it is set to `currentEnableNotification = c;` and goes under further checks using `doEnableNotification`.

However as this is happening in the first if else bracket ` enableNotificationQueue.add(c);` is never reached.

Steps to reproduce the behavior
-------------------------------
Should happen with an BLE device.

Do you have any idea what may have caused this?
-----------------------------------------------
Simple typo.

Do you have an idea how to solve the issue?
-------------------------------------------
Change 
```
  private void enableNotification(BluetoothGattCharacteristic c) {
    synchronized(enableNotificationQueue) {
      if (currentEnableNotification == null) {
        currentEnableNotification = c;
        if (!doEnableNotification(c))
          currentEnableNotification = null;
      } else
        enableNotificationQueue.add(c);
    }
  }
```
To 
```
  private void enableNotification(BluetoothGattCharacteristic c) {
    synchronized(enableNotificationQueue) {
      if (currentEnableNotification == null) {
        currentEnableNotification = c;
        if (!doEnableNotification(c))
          currentEnableNotification = null;
       else
        enableNotificationQueue.add(c);
      }
    }
  }
```

lordfolken, 2023-02-20:
Can you describe the symptom that you see and what it fixes?

ygaillard, 2023-02-20:
When `onServicesDiscovered`is called, which is used during scanning for BLE devices, all UUIDs are checked for a known service. This is for Flytec device 
```
 /* enable notifications for Flytec Sensbox */
    service = gatt.getService(BluetoothUuids.FLYTEC_SENSBOX_SERVICE);
    if (service != null) {
      BluetoothGattCharacteristic c =
        service.getCharacteristic(BluetoothUuids.FLYTEC_SENSBOX_NAVIGATION_SENSOR_CHARACTERISTIC);
      if (c != null) {
        setStateSafe(STATE_READY);
        enableNotification(c);
      }
```
When the service and characteristic are not null the enable notification is triggered. This is the routine which I mentioned above. As the new characteristic is not registered yet it result in null in the `currentEnableNotification` . 

For the current code on git `doEnableNotification(c)` is called and checked if it has valid descriptor (CCCD). If not `currentEnableNotification = null;` is executed but nothing further happens. `enableNotificationQueue.add(c);` is only called when the `currentEnableNotification` is already valid. However this can never happen, as `onServicesDiscovered` was already execute. And can only be executed again when reconnecting to the BLE device.

My correction suggest when `doEnableNotification` returns true the characteristic is added to the notification queue `enableNotificationQueue.add(c);`. And if it happen that this characteristic is already monitored. the `currentEnableNotification`should in my understanding return true, and no further adding to the notification queue is needed.

ygaillard, 2023-02-25:
Please ignore my issue, I had another own bug somewhere. The code is working perfectly fine.

lordfolken, 2023-02-25:
Thx for the heads up!

closed 2023-02-25 by lordfolken