XCSoar/XCSoar #1130: Some icons are worse than in v7.4

frank-vanderhulst, 2023-02-27:
XCSoar versions having and not having the problem
-------------------------------------------------
v7.8 has, v7.4 does not. I assume this change was made at v7.8

Expected behavior
-----------------
Whilst I accept that the icons needed to be reviewed, I think that some of the changes make XCSoar less usable. I think it's important to be able to take in information at a glance, and some of the new icons are larger and more complex, making that more difficult.

1. Mountain top (type 7) and Obstacle (type 8) have white backgrounds.

[image]

2. Normal waypoint (type 1) is now a faint crosshairs, making it much harder to pick out.

3. Obstacle (type 8) is now represented by what appears to me to be a transmitting tower. Previously it was just a simple A-frame tower, so suitable for other obstacles such as windmills.

4. Mountain top icon was previously a small black triangle, which made it ideal for point heights. Now it's a large icon with a snowcapped peak and a couple of subsidiary peaks, and visually similar to MtPass (type 6)
[image]

5. A couple of new icons (type 20 & 22) have been introduced. On close inspection, these appear to be hang gliders, but it isn't obvious at all... my first guess was satellite dishes. These are not documented. Oddly, there doesn't appear to be a type 21 icon.

6. Likewise, there are undocumented light blue icons for VOR & NDB (types 9 & 10) and type 19 - a light blue triangle in a circle.

Do you have an idea how to solve the issue?
-------------------------------------------
As I said, I welcome the effort to review and improve the icons. However, I would have been very happy if *additions* had been made to the existing set, rather than replacing some of them. I believe that there needs to be some rigor in documenting the set of icons available to developers, so that they can be used by all, and so that a change doesn't have a detrimental effect on pre-existing maps. So I think the solution is not in fixing this particular gripe, but in formalising a process for amending icons and assigning numbers to them.

A better solution would be to move the icons from the executable to separate e.g. .PNG files, so that a map developer could package up appropriate icons in the .XCM along with the topography Shapefiles.

lordfolken, 2023-02-27:
Please update your xcsoar to 7.28. Please download a current map. 
What platform are you using? Nothing on your screenshots looks like its supposed to...

frank-vanderhulst, 2023-02-27:
Hi,
Sorry, my mistake with the versions. I was using v7.24 and v7.28, not v7.4 and v7.8.

I am on v7.28. The screenshots are v7.28 on Windows.

The screenshots are of my own map, generated yesterday, so it is somewhat more current (and detailed) than what is available elsewhere. :) The standard maps do not have the spot heights, the windmills, the airstrips, or most of the roads. Many other features will appear quite different too, due to the extra detail I have.

Ah, but now I see that the white backgrounds I mentioned in point 1 are not visible on Android, so that must be an artefact of Windows. However, other icons (e.g. type 17 = Intersection) don't show the white background on Windows, so it must also be something to do with the definitions of those 2 icons.

Frank

lordfolken, 2023-02-27:
I cannot debug a map, that i don't know what has been done to it. What metadata standard it uses, nor even have access to. 

[image]

This is a blank XCSoar with 
 * ALPS_HighRes.xcm
 * CH.cup

running on windows 10. 

Can you rename your XCSoarData folder and download those things with the file manager and report back if you still see strange icons?

And if you do, i would need detailed information to what graphics card / driver version you use, because in the end this would be a rendering failure.

closed 2023-03-22 by lordfolken