XCSoar/XCSoar #1132: Plane Details, list selected data items not all copied

hjr, 2023-02-27:
XCSoar versions having and not having the problem
-------------------------------------------------
Head rev of today. Linux desktop, sim mode.

Expected behavior
-----------------
Choose a glider type from the internal polar list should copy all data into the plane profile.

Actual behavior
---------------
The selection misses to copy the handicap factor as well as the max. cruise speed. Because the "New" button shows the form prefilled with the current active data, this might slip through the users eye and results in a wrong plane data mix.

Steps to reproduce the behavior
-------------------------------
Config -> Plane easy and always to reproduce.

Do you have any idea what may have caused this?
-----------------------------------------------
The latest change/augmentation of the stored polar data in terms of masses might have missed out something.

Do you have an idea how to solve the issue?
-------------------------------------------
Yes

lordfolken, 2023-02-27:
This is all quite ugly.  In general new is new and shouldn't have any values defined. Copy as you state should copy. 

What works is, if you have configured cruise speed at for example at 275 and you copy a polar in with a predefined cruise speed then, it is set to that value. (Example AK-8) If its 0 then its not changed. 

What i'm confused about is the "Cruise speed" value. According to the manual (Section 14.7)  it should be km/h. But the value is clearly m/s.  For example AK-8 has a cruse speed of 50.0 which is then shown as 180 km/h.
The further 2 values are not described in the manual. 

In addition i noticed that the cruise speed value is capping any final glide calculations at that speed.  So auto mc in final glide will not exceed maneuvering speed, even though that may be desired.

hjr, 2023-02-27:
Hm, your last statement about the maneuvering speed is really interesting. I do not know the rational behind that.

PR #1140 (2023-03-09): Plane details, take over handicap factor from list selection. 
Brief summary of the changes
----------------------------
Copy one more item from the polar list selection.

Related issues and discussions
------------------------------
Closes #1132
closed 2023-03-09 by MaxKellermann