XCSoar/XCSoar #1133: our github CI chain seems broken

hjr, 2023-02-27:
XCSoar versions having and not having the problem
-------------------------------------------------
none, but the CI build env. My local build is fine.

Expected behavior
-----------------
build-macros should run through smooth and convert graphics etc. pp.

Actual behavior
---------------
The build-macros seem stop after this log line:
output/data/icons/winpilot_marginal_rgb.png output/data/icons/alt_landable_field_rgb.png output/data/icons/map_vor_160_rgb.png output/data/icons/settings_160_rgb.png

Steps to reproduce the behavior
-------------------------------
Trigger build of xcs head on github.

Do you have any idea what may have caused this?
-----------------------------------------------
No

Do you have an idea how to solve the issue?
-------------------------------------------
No

Scumi, 2023-02-27:
Its "build-macos" for Mac (not macros).

Issue is this:
`error: unknown warning option '-Wno-deprecated-experimental-coroutine' [-Werror,-Wunknown-warning-option]
`

lordfolken, 2023-02-27:
Max wrote the following:
https://github.com/XCSoar/XCSoar/blob/master/build/flags.mk#L7-L16

So someone should remove the flags try to compile with clang=y and fix any breakage.

lordfolken, 2023-02-27:
Found this commit at LLVM:
https://github.com/llvm/llvm-project/commit/55900a0d207670cb7295b98b31bb9ed50613cd42

Probably Max is using a git checkout of LLVM to compile. 

The clang supplied with github's osx machines is probably dated in comparison. 
The readme lists it as 14.0.0 and 15.0.4 available via brew:
https://github.com/actions/runner-images/blob/main/images/macos/macos-12-Readme.md

hjr, 2023-02-27:
Thx guys!

lordfolken, 2023-02-28:
@MaxKellermann what is the minimal version of clang required to build XCSoar? 
Debian seems to come with stone-age 11. Which we seem to be installing with the provision-scripts. 13 would be available from repositories.

fberst, 2023-03-02:
according to the developer manual, clang/LLVM 10 is the minimum version required,
but that doesn' work at the moment, i tried it in Debian with clang11 and even after disabling that option mentioned above i get a lot of errors. (concept / template related)....

It seems that clang11 doens't know std::invocable and std::predicate, i assume this libc++ version is too old.
No clue at the moment what to do about this......

the question is if we should stick to this acient clang version and work around this problems or raise the minimum compiler requirements. Debian 12 should be released later this year and will have clang-14, at this point we will raise the requirements.

hjr, 2023-03-03:
So, why not going with e.g. clang 12, or what ever the currently used standards covers?
I do not understand how the code using more recent standards then we agreed on made it into master, but that might be an unprofitable topic to discuss anyways.

lordfolken, 2023-03-03:
Rather than having a discussion about clang, i'd rather identify the particular commit that broke the CI toolchain. Can someone bisect?

fberst, 2023-03-04:
i have identified the problems and now i can compile it on debian with clang11....

1.  build/flags.mk: Wno-deprecated-experimental-coroutine unknown
2.  src/util/Concepts.hxx: std::invocable and std::predicate unknown, introduced in 7b1d120 ?
3.  src/Device/Driver/KRT2.cpp:  error: case value not in enumerated type 'std::byte'
4. linker error: /usr/bin/ld: cannot find -lc++abi, installing libc++abi-dev solves this

I will create commits to fix these, but i am no sure about the first 2 points.
1. Should we remove that flag in general? Should we check the compiler version and then decide if it's used?
2. Concepts.hxx has fallbacks for android / apple, add a check of the compiler version?

fberst, 2023-03-06:
at the moment i am out of ideas for this issue, after my last commits you can build it on debian with clang (if you disable -Wno-deprecated-experimental-coroutine).
on the github CI macos build it still fell apart (because sprintf is deprecated, which we use a lot).

and according to max we can't disable -Wno-deprecated-experimental-coroutine because this breaks his clang-14 build

closed 2023-03-21 by lordfolken