XCSoar/XCSoar #1134: max. cruise speed of polar

hjr, 2023-02-28:
XCSoar version
--------------
head/all so far

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
PoalrStore has a m/s representation of the max cruise speed, whereas other horizontal speeds are stored as km/h. Units are arbitrary, but the mix here is a bit awkward.
The polare store entry should use km/h also for the max cruise speed.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Get a better readability for the polar store data.

Any objection changing this?

lordfolken, 2023-02-28:
Interestingly the manual says km/h in the section 14.3.
This seems to be a winpilot polar standard, that is also importable via the polar dialog. We should align with that.

hjr, 2023-03-01:
Very good hint. The WinPilot plr file format, forgot about that. A quick research showed the WinPilot page is still up. Current manual (http://www.winpilot.com/UsersGuide/WinPilot_900.pdf) explains on page 82 the plr file format. There is no such thing like the max cruise speed. they show a similar example like the Ls3 in the XCS manual. So our manual I guess is wrong on this. The max cruise is a xcs extension on it and probably more a plane attribute rather than the polar description. 
Still the internal polar(plane) list has since quite some time more attributes than the plr file it might originate from. No reason to keep the relationship at all. The plr file import should be kept, why not.
I'd say .plr file format and internal PolarStore list format are two different topics.