XCSoar/XCSoar #1137: docker build for ANDROID target fails libsodium build/configure in ndk r25c

iltis42, 2023-03-03:
A recent git clone of XCSoar fails (2) now compiling for ANDROID target. 

According to the log ( ./output/ANDROID/armeabi-v7a/lib/build/libsodium-1.0.18/config.log ), clang binary is claimed to be missing (3), hence its found under a slightly different path (4), see below:
Searched:
/root/opt/android-ndk-r25**c**/toolchains/llvm/prebuilt/linux-x86_64/bin/clang
Located:
/root/opt/android-ndk-r25/toolchains/llvm/prebuilt/linux-x86_64/bin/clang

In targets.mk the ndk is defined as r25c, i get behind this if i change this to r25 only.
/build/targets.mk:  ANDROID_NDK ?= $(HOME)/opt/android-ndk-r25c

Can one of the android experts comment on this, do we need r25c, and why its failing ?

Thanks, Eckhard

Steps to reproduce:

git clone --recurse-submodules https://github.com/XCSoar/XCSoar XCSoar2;  cd XCSoar2;
docker build --no-cache --file ide/docker/Dockerfile -t xcsoar/xcsoar-build:latest ./ide/
docker run     --mount type=bind,source="$(pwd)",target=/opt/xcsoar  -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest /bin/bash xcsoar-compile ANDROID

2)
:
:
/* confdefs.h */
#define PACKAGE_NAME "libsodium"
#define PACKAGE_TARNAME "libsodium"
#define PACKAGE_VERSION "1.0.18"
#define PACKAGE_STRING "libsodium 1.0.18"
#define PACKAGE_BUGREPORT "https://github.com/jedisct1/libsodium/issues"
#define PACKAGE_URL "https://github.com/jedisct1/libsodium"
#define PACKAGE "libsodium"
#define VERSION "1.0.18"

configure: exit 77
Traceback (most recent call last):
  File "/opt/xcsoar/./build/thirdparty.py", line 210, in <module>
    x.build(toolchain)
  File "/opt/xcsoar/build/python/build/project.py", line 80, in build
    self._build(toolchain)
  File "/opt/xcsoar/build/python/build/autotools.py", line 109, in _build
    build = self.configure(toolchain, target_toolchain=target_toolchain)
  File "/opt/xcsoar/build/python/build/autotools.py", line 86, in configure
    subprocess.check_call(configure, cwd=build, env=toolchain.env)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/opt/xcsoar/output/src/libsodium-1.0.18/configure', 'CC=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin/clang', 'CXX=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++', 'CFLAGS=-Os -g -ffunction-sections -fdata-sections -fvisibility=hidden -mfloat-abi=softfp -mfpu=neon -fpic -funwind-tables -target armv7a-linux-androideabi21', 'CXXFLAGS=-Os -g -ffunction-sections -fdata-sections -fvisibility=hidden -mfloat-abi=softfp -mfpu=neon -fpic -funwind-tables -target armv7a-linux-androideabi21', 'CPPFLAGS=-isystem /opt/xcsoar/output/ANDROID/armeabi-v7a/lib/armv7a-linux-androideabi/include -DNDEBUG -I./output/ANDROID/include -DHAVE_POSIX -DHAVE_VASPRINTF -DANDROID -isystem ./output/ANDROID/armeabi-v7a/lib/armv7a-linux-androideabi/include -DEYE_CANDY -DXCSOAR_TESTING -DBIONIC -DLIBCPP_NO_IOSTREAM  ', 'LDFLAGS=-L/opt/xcsoar/output/ANDROID/armeabi-v7a/lib/armv7a-linux-androideabi/lib -Wl,--no-undefined -Wl,--fix-cortex-a8 -static-libstdc++ -lunwind ', 'LIBS= ', 'AR=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin//llvm-ar', 'ARFLAGS=rcs', 'RANLIB=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin//llvm-ranlib', 'STRIP=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin//llvm-strip', '--prefix=/opt/xcsoar/output/ANDROID/armeabi-v7a/lib/armv7a-linux-androideabi', '--disable-silent-rules', '--host=armv7a-linux-androideabi', '--disable-shared', '--enable-static']' returned non-zero exit status 77.
make[1]: *** [build/thirdparty.mk:41: output/ANDROID/armeabi-v7a/lib/stamp] Error 1
rm output/ANDROID/armeabi-v7a/dbg/src/Device/Driver/../dirstamp
make[1]: Leaving directory '/opt/xcsoar'
make: *** [build/android.mk:274: output/ANDROID/armeabi-v7a/dbg/bin/libxcsoar.so] Error 2
rm output/host/tools/GenerateSineTables/../dirstampmake: unlink: output/host/tools/GenerateSineTables/../dirstamp: Not a directory
 output/host/tools/GenerateSineTables.o

3)
/opt/xcsoar/output/src/libsodium-1.0.18/configure: line 3760: /root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin/clang: No such file or directory
[image]

4)
root@927a52758510:/# find . -name clang
./root/opt/android-ndk-r25/toolchains/llvm/prebuilt/linux-x86_64/share/clang
./root/opt/android-ndk-r25/toolchains/llvm/prebuilt/linux-x86_64/bin/clang
./root/opt/android-ndk-r25/toolchains/llvm/prebuilt/linux-x86_64/lib64/clang

fberst, 2023-03-03:
Not completely sure about this, i am no expert on docker.....

When you create the docker container it runs the script "/ide/provisioning/install-android-tools.sh" to download and setup the whole android-toolchain, so that you have the right version installed.

When the xcsoar sourcecode changes and requires a newer version you dont have installed (ndk r25c) it fails, as seen above.

To fix this you can re-run the install-android-tools.sh script inside your container.
The other alternative is to rebuild the container from scratch, takes longer and is quite unnecessary but should also work.

iltis42, 2023-03-03:
Thanks for the advice, in my steps to reproduce there is:  "docker build --no-cache --file ide/docker/Dockerfile -t xcsoar/xcsoar-build:latest ./ide/", so IMHO this won't do the trick here.

fberst, 2023-03-03:
my suggestions won't fix the problem, i build the container from scratch, ndk r25c ist installed, clang is found and then it still errors out while configuring libsodium.

`configure:3805: error: in /opt/xcsoar/output/ANDROID/armeabi-v7a/lib/build/libsodium-1.0.18`:
`configure:3807: error: C compiler cannot create executables`

something is not right, not quite sure what and why......

MaxKellermann, 2023-03-03:
Did you check config.log?

iltis42, 2023-03-03:
Yup, the screenshot above is the snippet from config log where it failed.

MaxKellermann, 2023-03-03:
That's because you have r25 instead of r25c.

fberst, 2023-03-03:
> Did you check config.log?

This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by libsodium configure 1.0.18, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ /opt/xcsoar/output/src/libsodium-1.0.18/configure CC=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin/clang CXX=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ CFLAGS=-Os -g -ffunction-sections -fdata-sections -fvisibility=hidden -fpic -funwind-tables -target i686-linux-android21 CXXFLAGS=-Os -g -ffunction-sections -fdata-sections -fvisibility=hidden -fpic -funwind-tables -target i686-linux-android21 CPPFLAGS=-isystem /opt/xcsoar/output/ANDROID/x86/lib/i686-linux-android/include -DNDEBUG -I./output/ANDROID/include -DHAVE_POSIX -DHAVE_VASPRINTF -DANDROID -D_BYTE_ORDER=_LITTLE_ENDIAN -isystem ./output/ANDROID/x86/lib/i686-linux-android/include -DEYE_CANDY -DXCSOAR_TESTING -DBIONIC -DLIBCPP_NO_IOSTREAM   LDFLAGS=-L/opt/xcsoar/output/ANDROID/x86/lib/i686-linux-android/lib -Wl,--no-undefined -static-libstdc++  LIBS=  AR=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin//llvm-ar ARFLAGS=rcs RANLIB=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin//llvm-ranlib STRIP=/root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin//llvm-strip --prefix=/opt/xcsoar/output/ANDROID/x86/lib/i686-linux-android --disable-silent-rules --host=i686-linux-android --disable-shared --enable-static

## --------- ##
## Platform. ##
## --------- ##

hostname = afec0cfdf756
uname -m = x86_64
uname -r = 5.10.0-20-amd64
uname -s = Linux
uname -v = #1 SMP Debian 5.10.158-2 (2022-12-13)

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2431: checking build system type
configure:2445: result: x86_64-pc-linux-gnu
configure:2465: checking host system type
configure:2478: result: i686-pc-linux-android
configure:2514: checking for a BSD-compatible install
configure:2582: result: /usr/bin/install -c
configure:2593: checking whether build environment is sane
configure:2648: result: yes
configure:2707: checking for i686-linux-android-strip
configure:2734: result: /root/opt/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/bin//llvm-strip
configure:2799: checking for a thread-safe mkdir -p
configure:2838: result: /bin/mkdir -p
configure:2845: checking for gawk
configure:2875: result: no
configure:2845: checking for mawk
configure:2861: found /usr/bin/mawk
configure:2872: result: mawk
configure:2883: checking whether make sets $(MAKE)
configure:2905: result: yes
configure:2934: checking whether make supports nested variables
configure:2951: result: yes
configure:3040: checking whether UID '0' is supported by ustar format
configure:3043: result: yes
configure:3050: checking whether GID '0' is supported by ustar format
configure:3053: result: yes
configure:3061: checking how to create a ustar tar archive
configure:3072: tar --version
tar (GNU tar) 1.34
[... 291 lines cut ...]
prefix='/opt/xcsoar/output/ANDROID/x86/lib/i686-linux-android'
program_transform_name='s,x,x,'
psdir='${docdir}'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target_alias=''
valgrind_enabled_tools=''
valgrind_tools=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "libsodium"
#define PACKAGE_TARNAME "libsodium"
#define PACKAGE_VERSION "1.0.18"
#define PACKAGE_STRING "libsodium 1.0.18"
#define PACKAGE_BUGREPORT "https://github.com/jedisct1/libsodium/issues"
#define PACKAGE_URL "https://github.com/jedisct1/libsodium"
#define PACKAGE "libsodium"
#define VERSION "1.0.18"

configure: exit 77

iltis42, 2023-03-03:
> That's because you have r25 instead of r25c.

Can't confirm that, in may targets.mk there is r25c

iltis42, 2023-03-03:
Hmm, i did a full and clean build from scratch IMHO, what's wrong with these steps:

git clone --recurse-submodules https://github.com/XCSoar/XCSoar XCSoar2 
cd XCSoar2
docker build --no-cache --file ide/docker/Dockerfile -t xcsoar/xcsoar-build:latest ./ide/
docker run --mount type=bind,source="$(pwd)",target=/opt/xcsoar -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest /bin/bash xcsoar-compile ANDROID

MaxKellermann, 2023-03-03:
I don't get it. You said you have r25, and you know it's a mistake to have r25 and not have r25c, still you attempt to start the build, knowing that it is doomed to fail. Then why do you discuss the build failure, instead of trying to find out why you don't have r25c?

(I've never used Docker and never will, it's too complicated for me. I'm always astounded by the multitude of problems people have with Docker. Building XCSoar is so easy, if you omit all the complexity that Docker is.)

iltis42, 2023-03-03:
Docker is a cool thing, it helps us to get such things done in one place and once its done everybody can benefit from that by just pull it. The steps i did should work in that way, and if not, we have a bug in there that needs a solution.
And nope, i did not say i have r25 when it fails, i see r25c when it fails in targets.mk.

MaxKellermann, 2023-03-03:
> And nope, i did not say i have r25 when it fails

You did say that. I don't get why you don't get the problem and why you repeat that stuff about targets.mk which makes no sense. I tried. So... good luck, I'm out of here.

iltis42, 2023-03-03:
This i said: "In targets.mk the ndk is defined as r25c, i get behind this if i change this to r25 only."

lordfolken, 2023-03-05:
> Hmm, i did a full and clean build from scratch IMHO, what's wrong with these steps:
> 
> git clone --recurse-submodules https://github.com/XCSoar/XCSoar XCSoar2 cd XCSoar2 docker build --no-cache --file ide/docker/Dockerfile -t xcsoar/xcsoar-build:latest ./ide/ docker run --mount type=bind,source="$(pwd)",target=/opt/xcsoar -it xcsoar/xcsoar-build:latest /bin/bash xcsoar-compile ANDROID

These steps result in an installable apk for me.

iltis42, 2023-03-05:
> These steps result in an installable apk for me.
Thank's for the update. Ran exactly the same batch again just now, and it's fine now. Must have been a temporary glitch in a packet that put somehow its binaries in the wrong place before. 

make[1]: Leaving directory '/opt/xcsoar'
  APK     output/ANDROID/dbg/build/unsigned.apk
  ALIGN   output/ANDROID/dbg/build/aligned.apk
  SIGN    output/ANDROID/bin/XCSoar-debug.apk

closed 2023-03-05 by iltis42

MaxKellermann, 2023-03-06:
> This i said: "In targets.mk the ndk is defined as r25c, i get behind this if i change this to r25 only."

Sigh. You said:

>  The issue there is that clang got placed here: /root/opt/android-ndk-r25/toolchains/llvm/prebuilt/linux-x86_64/bin/clang instead of ...r25c/...

This clearly says you have r25 and not r25c.

Of course, then changing the XCSoar build system to use r25 instead of r25c, you can make it work, duh!
But still it's only because you have r25 and not r25c.

This "temporary glitch" is not some unexplainable mystery - it's a simple and deterministic problem: you had installed r25 instead of r25c!

iltis42, 2023-03-06:
@lordfolken strange, afaik the android ndk version comes with the docker yml, or is there something referenced outside the docker ?