XCSoar/XCSoar #1142: USE_FB=y fails to compile current head

lordfolken, 2023-03-05:
XCSoar versions having and not having the problem
-------------------------------------------------
Current head

Expected behavior
-----------------

It should compile and result in a binary that displays on the framebuffer without further 

Actual behavior
---------------
```bash
$ make -j$(nproc) TARGET=UNIX USE_CCACHE=y USE_FB=y
```

```bash
src/MainWindow.cpp: In member function ‘void MainWindow::Initialise()’:
src/MainWindow.cpp:175:3: error: ‘SetCursorSize’ was not declared in this scope
  175 |   SetCursorSize(CommonInterface::GetDisplaySettings().cursor_size);
      |   ^~~~~~~~~~~~~
src/MainWindow.cpp:176:3: error: ‘SetCursorColorsInverted’ was not declared in this scope
  176 |   SetCursorColorsInverted(CommonInterface::GetDisplaySettings().invert_cursor_colors);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
make: *** [build/compile.mk:108: output/UNIX/dbg/src/MainWindow.o] Error 1
make: *** Waiting for unfinished jobs....

```

Steps to reproduce the behavior
-------------------------------
See above

Do you have any idea what may have caused this?
-----------------------------------------------

Probably a misplaced #ifdef o no FB specific implementation of SetCursorSize
I would have to investigate if this is an x86 specific failure. 

Do you have an idea how to solve the issue?
-------------------------------------------
Yes, but i need time for an analysis. 

In general XCSoar probably should add USE_FB to the compile chain, as we have  a few platforms that depend on FB, such as raspberry pis and openvario.

lordfolken, 2023-03-10:
@linuxianer99 can you compile the current head for openvario?

fberst, 2023-03-10:
I investigated this issue a little bit, and the cause for this problem is that in "src/ui/window/TopWindow.hpp" DRAW_MOUSE_CURSOR is not defined -> the declarations for this functions are missing.

in TopWindow.hpp DRAW_MOUSE_CURSOR gets usually defined in "ui/opengl/Features.hpp" (if opengl is enabled)
The FB build doesn't use opengl -> so it is missing

I was able to fix it by including "ui/canvas/Features.hpp" in TopWindow.hpp
But i would be surprised if this doesn't break something else, we have a lot of build targets and options, i have to further test this.

linuxianer99, 2023-03-14:
> @linuxianer99 can you compile the current head for openvario?

I give it a try ... build is currently running ...

lordfolken, 2023-03-22:
@MaxKellermann could you help here? This looks like a misplaced ifdef...

closed 2023-04-29 by MaxKellermann