XCSoar/XCSoar #1143: Airspace File Error

Ikarus55, 2023-03-05:
Which XCSoar version are you using when seeing the problem?
-------------------------------

XCSoar for Android 7.26 and newer

Which XCSoar version is the last one you know that did not show the problem?
-------------------------------
7.24

Actual behavior:
-------------------------------
When starting XCSoar, there is an error with regards to the airspace file, see first screenshot. As a consequence, no airspace is shown on the map. 

When opening the airspace file in editor, the top altitude is available, second screenshot.

Steps to reproduce the behavior
-------------------------------
I can reproduce the error with other Android devices, using XCSoar 7.28, and other airspace files. For example with the New Zealand airspace file which I directly downloaded through XCSoar ->Config ->System ->Site Files -> Airspace. Again, upon startup i get an error regarding missing top altitude in a random airspace.

Do you have any idea what may have caused this?
-----------------------------------------------

no, the airspace file worked fine for versions 7.24 and older

[image]

[image]

fberst, 2023-03-05:
Can you please upload your airspace-file that causes the problem here?
I can't reproduce the problem, New_Zealand_Airspace.txt downloads and opens without any error here.

Ikarus55, 2023-03-05:
Sure! Attached the airspace file. I 'downgraded' to version 7.24 on my device and it works well.

[Airspace_Balazs_Mar2022_short.txt](https://github.com/XCSoar/XCSoar/files/10893209/Airspace_Balazs_Mar2022_short.txt)

lordfolken, 2023-03-05:
Thx for the file. 

XCSoar's airspace parser is now less forgiving of errors and will report them to you. 
7.24 Lulled you in a false sense of security: it will just not render the airspace, potentially not warning you of violation.

The error is actually in the airspace before the one you mentioned. Why the error message is misleading i can't say. 
Fixing that one yields a further error in the "Axes de Voltige" beginning 24554. There DB is used to try to draw lines with only 2 coordinates. 

According to http://winpilot.com/usersguide/userairspace.asp DB is a polygon and therefore needs at least 3 coordinates.

Btw i added openaip airspaces to the repository (file manager) you can download airspaces for all the countries in your file there.

lordfolken, 2023-03-05:
I'm creating a bug report for the errornous error message. But its still your airspace file that needs fixing.

Ikarus55, 2023-03-06:
many thanks for your help!